P1sec/pycrate #121: How to reset a class to it's empty default state

vazir, 2020-11-07:
Hello! 
Is there any way to reset a class to empty state? When first loaded, it has empty structure
like 
TCAP_MAPv2v3.TCAP_MAP_Messages.TCAP_MAP_Message()
gives
None
and obviously after initializing with any from_ber or whatever at gets the content, so how to clean it?
Is there any root object which can be set to None or method, cleaning up everything?

p1-bmu, 2020-11-09:
When encoding / decoding complex objects, you set values into internal components, not only into the object itself. So, setting `None` in the object's value is not a complete clean-up.

There are 2 solutions to do so:
1) You can `deepcopy()` the ASN.1 object you want to use for encoding / decoding: each time you do so, you have an empty and clean object. This can however be quite costly, especially for big object such as TCAP-MAP-Message.
2) You can call the `reset_val()` method of the object. This will clean-up all internal values and components.
https://github.com/P1sec/pycrate/blob/da5afc091ebae21674908d90a7a9584eca4b0f69/pycrate_asn1rt/asnobj.py#L1105

vazir, 2020-11-09:
Thank you, it's what i have been looking for!
deepcopy, (softly speaking), is a heavy operation, so heavy so it's almost unusable for production. It's much more feasible to use one-pass init, generate and than cleanup :) Cleanup with reset_val() and re-initializing from the object() structure is a WAY faster than deepcopying. Will the object do cleanup itself, before
Below are some speedtests, including deeptopy of TCAP_CAP IDP message 

Deepcopy of the clean TCAP_CAP.TCAP_CAP_Messages.TCAP_CAP_Message
Total func Deepcopy of the clean TCAP_CAP.TCAP_CAP_Messages.TCAP_CAP_Message - deepcopy time taken for 1000 iterations is: 100.030975, and 0.100031 per iteration which is 10 iters per second.
Testing decoding from bytes
Total func Testing decoding from bytes - from_ber time taken for 1000 iterations is: 0.321923, and 0.000322 per iteration which is 3106 iters per second.
Testing appying from object
Total func Testing appying from object - set_val time taken for 1000 iterations is: 0.069550, and 0.000070 per iteration which is 14378 iters per second.
Testing encoding to bytes
Total func Testing encoding to bytes - to_ber time taken for 1000 iterations is: 0.398271, and 0.000398 per iteration which is 2511 iters per second.
Testing conversion to json
Total func Testing conversion to json - to_jer time taken for 1000 iterations is: 0.148707, and 0.000149 per iteration which is 6725 iters per second.
Testing conversion from json
Total func Testing conversion from json - from_jer time taken for 1000 iterations is: 0.129513, and 0.000130 per iteration which is 7721 iters per second.
Testing getting object
Total func Testing getting object - (NoName) time taken for 1000 iterations is: 0.000173, and 0.000000 per iteration which is 5769331 iters per second.
Testing pickling dump
Total func Testing pickling dump - dumps time taken for 1000 iterations is: 0.003340, and 0.000003 per iteration which is 299401 iters per second.
Decode BER and encode JSON
Total func Decode BER and encode JSON - complex_decode time taken for 1000 iterations is: 9.023237, and 0.009023 per iteration which is 111 iters per second.
Decode JSON and encode BER
Total func Decode JSON and encode BER - complex_encode time taken for 1000 iterations is: 9.130442, and 0.009130 per iteration which is 110 iters per second.
Decode BER and2Pickle
Total func Decode BER and2Pickle - complex_dec_pickle time taken for 1000 iterations is: 8.861933, and 0.008862 per iteration which is 113 iters per second.
Unpickle and encode BER
Total func Unpickle and encode BER - complex_enc_unpickle time taken for 1000 iterations is: 9.259791, and 0.009260 per iteration which is 108 iters per second.
Objects equality: True

p1-bmu, 2020-11-10:
Basically, in a single thread environment, you don't need to clean-up any ASN.1 objects and can use them for several sequential encoding / decoding as such.
In case you are willing to do multi-threading, then having a deepcopy of each object per thread seems a good idea to me.

closed 2021-10-21 by p1-bmu