P1sec/pycrate #123: Accessing message name for RRC messages

aveer28, 2020-11-11:
I'm decoding RRC messages as follows

```
defs = RRCLTE.EUTRA_RRC_Definitions
msg = defs.BCCH_DL_SCH_Message

msg.from_uper(unhexlify(input))
decoded = msg._val
```
Is there any way to access the message name e.g. measurementControl from the `decoded` or other object.

p1-bmu, 2020-11-12:
Sorry, but I cannot really understand your question. Please show what's inside your `decoded` dict, and explain what _other object_ you are considering.

aveer28, 2020-11-13:
Here is an example of the decoded dict. 

```
{'message': ('c1', ('systemInformationBlockType1', {'cellAccessRelatedInfo': {'plmn-IdentityList': [{'plmn-Identity': {'mcc': [2, 3, 4], 'mnc': [1, 0]}, 'cellReservedForOperatorUse': 'reserved'}, {'plmn-Identity': {'mcc': [2, 3, 4], 'mnc': [1, 5]}, 'cellReservedForOperatorUse': 'notReserved'}], 'trackingAreaCode': (8254, 16), 'cellIdentity': (128062750, 28), 'cellBarred': 'notBarred', 'intraFreqReselection': 'allowed', 'csg-Indication': False}, 'cellSelectionInfo': {'q-RxLevMin': -62}, 'p-Max': 23, 'freqBandIndicator': 20, 'schedulingInfoList': [{'si-Periodicity': 'rf16', 'sib-MappingInfo': []}, {'si-Periodicity': 'rf16', 'sib-MappingInfo': ['sibType3']}, {'si-Periodicity': 'rf16', 'sib-MappingInfo': ['sibType5']}, {'si-Periodicity': 'rf16', 'sib-MappingInfo': ['sibType6']}, {'si-Periodicity': 'rf16', 'sib-MappingInfo': ['sibType7']}], 'si-WindowLength': 'ms20', 'systemInfoValueTag': 16}))}
```

For me to access the message name `systemInformationBlockType1` or other params, I would have to step down through the dict and tuples. I was wondering if there was an easier way to access this, as it appears for the NAS decoder, which creates objects corresponding to the different message types. I realise that NAS is more advanced and thus this functionality may not exist.

p1-bmu, 2020-11-20:
As shortly introduced in https://github.com/P1sec/pycrate/wiki/Using-the-pycrate-asn1-runtime#rrc-3g, you can use `get_at()` and `get_val_at()` methods to navigate within an object or its value. This however does not save you to know the object content, which only depends of the compiled ASN.1 schema, and nothing else.

closed 2020-11-26 by aveer28