P1sec/pycrate #124: Encoding asn1 format string

mamkarim, 2020-11-12:
Hi,
I am trying to encode asn1 format string to dict object, but getting empty value. I tried to find a solution in the wiki and issues, but no luck. Any help will be very much appreciated. Below is my code -

from pycrate_asn1rt.utils import * 
from binascii import unhexlify, hexlify
from pycrate_asn1dir import RRC3G

pcch = RRC3G.Class_definitions.PCCH_Message
pcch.from_uper(unhexlify('4455c803999055c601b95855aa06b09e'))

test = pcch.from_asn1(pcch.to_asn1())
print('test',test)
              
print(pcch.set_val(test))

p1-bmu, 2020-11-12:
Maybe you should exercise as proposed in the wiki first: https://github.com/P1sec/pycrate/wiki/Using-the-pycrate-asn1-runtime.
As you can see from here, and the source code and the doc strings, neither methods `from_asn1()` nor `set_val()` returns ; therefore you will only get `None` printed...

mamkarim, 2020-11-12:
Thanks for the hint. understood. Much appreciated.

from pycrate_asn1rt.utils import *
from binascii import unhexlify, hexlify
from pycrate_asn1dir import RRC3G

pcch = RRC3G.Class_definitions.PCCH_Message
pcch.from_uper(unhexlify('4455c803999055c601b95855aa06b09e'))

pcch.from_asn1(pcch.to_asn1())
test = pcch()
print('test',test)

closed 2020-11-12 by mamkarim