P1sec/pycrate #126: RRCLTE to_json in Python 2.7

hektordaniel, 2020-11-29:
Hi all

I have two servers running decoding of RRC LTE. However one server is python 2.7 and other server is python 3.8.  Runnning following code in python 2.7 is getting error, beside in python 3.8 environments run well. It looks that in python 2.7 is not recognizing the functions related to JER and JSON:

>>> import json
>>> from pycrate_asn1dir import RRCLTE
>>> from pycrate_asn1rt.utils import *
>>> ULCCH = RRCLTE.EUTRA_RRC_Definitions.UL_CCCH_Message
>>> MSG=unhexlify("074172083900")
>>> ULCCH.from_uper(MSG)
>>> stringdecoded=ULCCH.to_json()
>>> print stringdecoded
{
  message c1 : rrcConnectionReestablishmentRequest : {
    criticalExtensions rrcConnectionReestablishmentRequest-r8 : {
      ue-Identity {
        c-RNTI '3A0B'H -- :
                            --,
        physCellId 288,
        shortMAC-I '8390'H
      },
      reestablishmentCause reconfigurationFailure,
      spare '00'B
    }
  }
}

>>> dd=ULCCH.to_json()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'SEQ' object has no attribute 'to_json'

if I check the functions for the class ULCCH in different versions of python can be observed that JSON and JER functions are not recognized by python2.7. Could you please confirm me if there is any workaround for this, different to upgrade python?

PYTHON 3.8
dir(ULCCH)
['CLASS', 'DEFAULT_TRANS', 'ENV_SEL_TRANS', 'TAG', 'TYPE', '_ASN1Obj__to_ber_codec_set', '_ASN1Obj__to_ber_codec_unset', '_SAFE_BND', '_SAFE_BNDTAB', '_SAFE_DYN', '_SAFE_INIT', '_SAFE_STAT', '_SAFE_VAL', '_SILENT', '__bin__', '__bytes__', '__call__', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__hex__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_chk_hier', '_chk_trans', '_const_tab', '_const_val', '_cont', '_cont_tags', '_decode_ber_cont', '_decode_ber_cont_ws', '_def', '_encode_ber_cont', '_encode_ber_cont_ws', '_env', '_ext', '_from_asn1', '_from_ber', '_from_ber_ws', '_from_jval', '_from_jval_wrap', '_from_oer', '_from_oer_ws', '_from_per', '_from_per_ws', '_get_obj_by_path', '_get_tab_obj', '_get_tab_obj_nonuniq', '_get_tab_obj_uniq', '_get_val_by_path', '_group', '_hier', '_log', '_mod', '_mode', '_name', '_opt', '_param', '_parent', '_root', '_root_mand', '_root_opt', '_safechk_bnd', '_safechk_obj', '_safechk_set', '_safechk_set_int', '_safechk_set_real', '_safechk_set_str', '_safechk_val', '_safechk_val_int', '_safechk_val_real', '_safechk_val_str', '_safechk_valcompl', '_tag', '_tagc', '_to_asn1', '_to_ber', '_to_ber_ws', '_to_jval', '_to_jval_wrap', '_to_oer', '_to_oer_ws', '_to_per', '_to_per_ws', '_tr', '_trans', '_transauto', '_typeref', '_uniq', '_val', 'bin', 'convert_named_val', 'dec_hier', 'from_aper', 'from_aper_ws', 'from_asn1', 'from_ber', 'from_ber_ws', 'from_bytes', 'from_cer', 'from_cer_ws', 'from_coer', 'from_coer_ws', 'from_der', 'from_der_ws', 'from_gser', 'from_int', 'from_jer', 'from_json', 'from_oer', 'from_oer_ws', 'from_uint', 'from_uper', 'from_uper_ws', 'fullname', 'get_at', 'get_complexity', 'get_const', 'get_env', 'get_header', 'get_hier', 'get_hier_abs', 'get_internals', 'get_len', 'get_next', 'get_payload', 'get_prev', 'get_proto', 'get_root', 'get_root_path', 'get_trans', 'get_type_list', 'get_typeref', 'get_typeref_list', 'get_val', 'get_val_at', 'get_val_paths', 'hex', 'in_class', 'inc_hier', 'iter_cont', 'reautomate', 'reset_val', 'set_env', 'set_hier', 'set_trans', 'set_transauto', 'set_val', 'set_val_at', 'show', 'to_aper', 'to_aper_ws', 'to_asn1', 'to_ber', 'to_ber_ws', 'to_bytes', 'to_cer', 'to_cer_ws', 'to_coer', 'to_coer_ws', 'to_der', 'to_der_ws', 'to_gser', 'to_int',
 'to_jer', 'to_json', 
'to_oer', 'to_oer_ws', 'to_uint', 'to_uper', 'to_uper_ws', 'unset_val']

PYTHON2.7
dir(ULCCH)
['CLASS', 'DEFAULT_TRANS', 'ENV_SEL_TRANS', 'TAG', 'TYPE', '_ASN1Obj__to_ber_codec_set', '_ASN1Obj__to_ber_codec_unse        t', '_SAFE_BND', '_SAFE_BNDTAB', '_SAFE_DYN', '_SAFE_INIT', '_SAFE_STAT', '_SAFE_VAL', '_SILENT', '__bin__', '__call_        _', '__class__', '__delattr__', '__dict__', '__doc__', '__format__', '__getattribute__', '__hash__', '__hex__', '__in        it__', '__module__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '_        _subclasshook__', '__weakref__', '_chk_hier', '_chk_trans', '_const_tab', '_const_val', '_cont', '_cont_tags', '_deco        de_ber_cont', '_decode_ber_cont_ws', '_def', '_encode_ber_cont', '_encode_ber_cont_ws', '_env', '_ext', '_from_asn1',         '_from_ber', '_from_ber_ws', '_from_oer', '_from_oer_ws', '_from_per', '_from_per_ws', '_get_obj_by_path', '_get_tab        _obj', '_get_tab_obj_nonuniq', '_get_tab_obj_uniq', '_get_val_by_path', '_group', '_hier', '_log', '_mod', '_mode', '        _name', '_opt', '_param', '_parent', '_root', '_root_mand', '_root_opt', '_safechk_bnd', '_safechk_obj', '_safechk_se        t', '_safechk_set_int', '_safechk_set_real', '_safechk_set_str', '_safechk_val', '_safechk_val_int', '_safechk_val_re        al', '_safechk_val_str', '_safechk_valcompl', '_tag', '_tagc', '_to_asn1', '_to_ber', '_to_ber_ws', '_to_oer', '_to_o        er_ws', '_to_per', '_to_per_ws', '_tr', '_trans', '_transauto', '_typeref', '_uniq', '_val', 'bin', 'convert_named_va        l', 'dec_hier', 'from_aper', 'from_aper_ws', 'from_asn1', 'from_ber', 'from_ber_ws', 'from_bytes', 'from_cer', 'from_        cer_ws', 'from_coer', 'from_coer_ws', 'from_der', 'from_der_ws', 'from_gser', 'from_int', 'from_oer', 'from_oer_ws',         'from_uint', 'from_uper', 'from_uper_ws', 'fullname', 'get_at', 'get_complexity', 'get_const', 'get_env', 'get_header        ', 'get_hier', 'get_hier_abs', 'get_internals', 'get_len', 'get_next', 'get_payload', 'get_prev', 'get_proto', 'get_r        oot', 'get_root_path', 'get_trans', 'get_type_list', 'get_typeref', 'get_typeref_list', 'get_val', 'get_val_at', 'get        _val_paths', 'hex', 'in_class', 'inc_hier', 'reautomate', 'reset_val', 'set_env', 'set_hier', 'set_trans', 'set_trans        auto', 'set_val', 'set_val_at', 'show', 'to_aper', 'to_aper_ws', 'to_asn1', 'to_ber', 'to_ber_ws', 'to_bytes', 'to_cer', 'to_cer_ws', 'to_coer', 'to_coer_ws', 'to_der', 'to_der_ws', 'to_gser', 'to_int', 'to_oer', 'to_oer_ws', 'to_uint        ', 'to_uper', 'to_uper_ws', 'unset_val']

p1-bmu, 2020-11-30:
Thanks for the feedback. As you should know, Python 2 is end of life, and should not be used anymore.
I checked on my side and found that some evolution of Python 2.7 (now 2.7.18...) lead to broken features in pycrate: support for JSONDecoderError, certain kind of method calls, and representation of scientific numbers... I fixed all of them, so everyone can again perfectly enjoy pycrate with Python 2.7 !
But I can only recommend you to move to Python 3.

hektordaniel, 2020-12-01:
Ok thanks for your help. I am going to upgrade to python 3.

closed 2020-12-07 by p1-bmu