P1sec/pycrate #128: decoding of DUtoCURRCContainer F1AP msg (InitialULRRCMessageTransfer)

DevD3339, 2021-01-09:
Hi, I am trying to decode the DUtoCURRCContainer of F1AP msg (InitialULRRCMessageTransfer), raises an error of bitlen overflow. 

Could anyone help me to resolve the issue and point out where am I going wrong.

Thanks

Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pycrate_asn1dir.F1AP import *
>>> from pycrate_asn1rt.utils import *
>>> from pycrate_asn1dir import RRCNR
>>> from binascii import hexlify, unhexlify
>>> initial_ul_rrc_msg = '000b408085000005002900020001006f00090013f1840000000010005f0003000064003200070610000000006700800059585c40b001100aec81d061ea007c22a09507b8c0207a980cb200654004400000000078038000010dc2108003f88c90810120301a80098aa84400000192100001610c6f28101004020004000000000000000000180101c08a80'
>>> initial_ul_rrc_msg_pdu = F1AP_PDU_Descriptions.F1AP_PDU
>>> rrc_container_pdu = RRCNR.NR_RRC_Definitions.UL_CCCH_Message
>>> initial_ul_rrc_msg_pdu.from_aper(unhexlify(initial_ul_rrc_msg))
>>> 
>>> initial_ul_rrc_msg_pdu()
('initiatingMessage', {'procedureCode': 11, 'criticality': 'ignore', 'value': ('InitialULRRCMessageTransfer', {'protocolIEs': [{'id': 41, 'criticality': 'reject', 'value': ('GNB-DU-UE-F1AP-ID', 1)}, {'id': 111, 'criticality': 'reject', 'value': ('NRCGI', {'pLMN-Identity': b'\x13\xf1\x84', 'nRCellIdentity': (1, 36)})}, {'id': 95, 'criticality': 'reject', 'value': ('C-RNTI', 100)}, {'id': 50, 'criticality': 'reject', 'value': ('RRCContainer', b'\x10\x00\x00\x00\x00g')}, {'id': 128, 'criticality': 'reject', 'value': ('DUtoCURRCContainer', b'\\@\xb0\x01\x10\n\xec\x81\xd0a\xea\x00|"\xa0\x95\x07\xb8\xc0 z\x98\x0c\xb2\x00e@\x04@\x00\x00\x00\x00x\x03\x80\x00\x01\r\xc2\x10\x80\x03\xf8\x8c\x90\x81\x01 0\x1a\x80\t\x8a\xa8D\x00\x00\x01\x92\x10\x00\x01a\x0co(\x10\x10\x04\x02\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x18\x01\x01\xc0\x8a\x80')}]})})
>>> 
>>> du_to_cu_rrc = F1AP_IEs.DUtoCURRCContainer
>>> du_to_cu_rrc_pdu = get_val_at(initial_ul_rrc_msg_pdu, ['initiatingMessage','value','InitialULRRCMessageTransfer','protocolIEs'])[4]['value'][1]
>>>
>>> print(du_to_cu_rrc.from_aper(du_to_cu_rrc_pdu))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/pycrate-0.4-py3.6.egg/pycrate_asn1rt/asnobj.py", line 1311, in from_aper
    self._from_per(char)
  File "/usr/local/lib/python3.6/dist-packages/pycrate-0.4-py3.6.egg/pycrate_asn1rt/asnobj_str.py", line 1461, in _from_per
    self.__from_per(char, unconst=True)
  File "/usr/local/lib/python3.6/dist-packages/pycrate-0.4-py3.6.egg/pycrate_asn1rt/asnobj_str.py", line 1490, in __from_per
    self._val = ASN1CodecPER.decode_unconst_open(char)
  File "/usr/local/lib/python3.6/dist-packages/pycrate-0.4-py3.6.egg/pycrate_asn1rt/codecs.py", line 860, in decode_unconst_open
    return char.get_bytes(8*ldet)
  File "/usr/local/lib/python3.6/dist-packages/pycrate-0.4-py3.6.egg/pycrate_core/charpy.py", line 376, in get_bytes
    .format(bitlen, self._len_bit-self._cur)))
pycrate_core.charpy.CharpyErr: bitlen overflow: 736, max 696
>>>
>>> print(du_to_cu_rrc.from_uper(du_to_cu_rrc_pdu))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/pycrate-0.4-py3.6.egg/pycrate_asn1rt/asnobj.py", line 1277, in from_uper
    self._from_per(char)
  File "/usr/local/lib/python3.6/dist-packages/pycrate-0.4-py3.6.egg/pycrate_asn1rt/asnobj_str.py", line 1461, in _from_per
    self.__from_per(char, unconst=True)
  File "/usr/local/lib/python3.6/dist-packages/pycrate-0.4-py3.6.egg/pycrate_asn1rt/asnobj_str.py", line 1490, in __from_per
    self._val = ASN1CodecPER.decode_unconst_open(char)
  File "/usr/local/lib/python3.6/dist-packages/pycrate-0.4-py3.6.egg/pycrate_asn1rt/codecs.py", line 860, in decode_unconst_open
    return char.get_bytes(8*ldet)
  File "/usr/local/lib/python3.6/dist-packages/pycrate-0.4-py3.6.egg/pycrate_core/charpy.py", line 376, in get_bytes
    .format(bitlen, self._len_bit-self._cur)))
pycrate_core.charpy.CharpyErr: bitlen overflow: 736, max 696

p1-bmu, 2021-01-11:
As you can see from the F1AP specification, the DUtoCURRCContainer is just an OCTET STRING:
```
In [7]: F1AP_IEs.DUtoCURRCContainer                                                                                                                       
Out[7]: <DUtoCURRCContainer (OCTET STRING)>
```

This probably means that the DU is not required to deal with what's inside the container. If you want to decode the content of it, you need to read the 3GPP TS 38.473, section 9.2.3.1: you may need some structures from the NR RRC specifications, such as:
```
In [8]: from pycrate_asn1dir.RRCNR import *                                                                                                               
In [9]: NR_RRC_Definitions.CellGroupConfig
```

Take care as all RRC messages are UPER encoded, and not APER.

DevD3339, 2021-01-11:
Thanks, it worked. 

Should I open another question? 

Another question is - when we replay the binary data of any F1AP msg, it decodes perfectly, but when we replay the IEs by building the complete msg or just substituting like val = ('initiatingMessage', {'procedureCode': 1, 'criticality': 'ignore', 'value' : ('F1SetupRequest', {'protocolIEs': **IEs}**)}) it throws ASN.1 raise(ASN1ObjErr('{0}: invalid value, {1!r}'.format(self.fullname(), val)))

IEs which we decoded earlier and extracted.

p1-bmu, 2021-01-11:
If it says "invalid value", it means the value you built is invalid.

DevD3339, 2021-01-11:
OK, but the msg was decoded, and rebuilt the msg with the extracted value. NO manipulation in between. Any ideas how can we resolve.

p1-bmu, 2021-01-14:
Please provide the code that triggers this issue, otherwise I cannot help you.