P1sec/pycrate #134: Question: NASLTE:How to encode JSON to bytes

renmnair, 2021-03-24:
Hi,

I have this query regarding encoding json to NAS pdu ,

I have the below NAS pdu for Attach accept message which I decoded using the parse_NAS_MT function

hexstr = "17af5e6481020742015f060010f2430002004d5201c101091b0764656661756c74066d6e63303334066d636330313204677072730501c0a80202917b00228080210a0300000a810608080808000d040808080800160400000000001903000000500bf610f2438001014c166a1e3408031f19f1031f11f26402810af25e01266a0126"
.

Msg, err = parse_NAS_MT(unhexlify(hex_string))

then I changed few values after converting the Msg to json.

Msg_json = Msg.to_json()

Is it possible to encode the **new json** (after i did change values to Msg_json) to bytes?

Thanks,
Renjana

p1-bmu, 2021-03-24:
You should be able to do this:
```
Msg.from_json( Msg_json )
```

renmnair, 2021-03-28:
Thanks..that worked.

I have one more question, 
When trying decode the hex string from dedicatedNasinfo of the Attach accept message,
```
 hexstr = "27af5e6481020742015f060010f2430002004d5201c101091b0764656661756c74066d6e63303334066d636330313204677072730501c0a80202917b00228080210a0300000a810608080808000d040808080800160400000000001903000000500bf610f2438001014c166a1e3408031f19f1031f11f26402810af25e01266a0126" 
```   
[image]

I do not get the full decoded Msg,
```
In [10]: Msg                                                                                                                                                                                                                                 
Out[10]: <EMMSecProtNASMessage : <EMMHeaderSec : <SecHdr : 2 (Integrity protected and ciphered)><ProtDisc : 7 (EMM)>><MAC : 0xaf5e6481><Seqn : 2><NASMessage : b'\x07B\x01_\x06\x00\x10\xf2C\x00\x02\x00MR\x01\xc1\x01\t\x1b\x07default\x06mnc034\x06mcc012\x04gprs\x05\x01\xc0\xa8\x02\x02\x91{\x00"\x80\x80!\n\x03\x00\x00\n\x81\x06\x08\x08\x08\x08\x00\r\x04\x08\x08\x08\x08\x00\x16\x04\x00\x00\x00\x00\x00\x19\x03\x00\x00\x00P\x0b\xf6\x10\xf2C\x80\x01\x01L\x16j\x1e4\x08\x03\x1f\x19\xf1\x03\x1f\x11\xf2d\x02\x81\n\xf2^\x01&j\x01&'>>
```
I see this when the SecHdr is 2
When security header is 1, then i get a full decode.  
Is this intended..?

p1-bmu, 2021-03-29:
yes, as you can read: `<SecHdr : 2 (Integrity protected and ciphered)>`
It's expected to be encrypted.