P1sec/pycrate #144: move various print() to a proper logging mechanism

p1-bmu, 2021-05-21:
Various `print()` exists throughout pycrate, that should be moved to a proper logging mechanism, in order to not eventually pollute stdout. All logging should be done with the `log()` function in `./pycrate_core/utils.py`, and this log function should do a proper logging instead of just a print.

Places where explicit `print()` should be changed to `log()`:
- ./pycrate_ether/IP.py

benmaddison, 2021-05-22:
> Various `print()` exists throughout pycrate, that should be moved to a proper
> logging mechanism, in order to not eventually pollute stdout. All logging
> should be done with the `log()` function in `./pycrate_core/utils.py`, and
> this log function should do a proper logging instead of just a print.

It's more work, but better to call `log(...)` directly in the module where the event occurs.

The reason for this is with the boilerplate (in every module):

``` python
import logging

log = logging.getLogger(__name__)
```

Then the `log` object will have a name matching the name of the module, which
makes it easy to see where in the library the message was generated.

If you centralise the call, everything will look like it came from `pycrate_core.utils`

p1-bmu, 2021-05-25:
Yes, I realize this while reading the `logging` module doc and tested it.
This will require changes in many files : I'll do it when I a have proper time.