P1sec/pycrate #148: Need sample example for GTP-Cv2  encoding

akibsayyed, 2021-06-25:
I am using pycrate for internal study and research. I am trying to use pycreate for encoding GTP-Cv2.

I am able to perform echo request and response but developing further is bit complicated.Can you please guide me how to use pycrate for GTP-Cv2 encoding with short example.

p1-bmu, 2021-06-25:
I just pushed a commit that provide a way to initialize IEs in GTP-C messages. This is however far from automatically setting well-formed message, and there is unfortunately no automatic or easy way to set proper GTP-C messages without reading through the TS 29.274 specification. When you set a GTP-C message, you need to know which optional IEs are required for your need, and you need to set appropriate values into those IEs according to their specific structures.

There is the method `init_ies()` that you can use to initialize all mandatory (plus eventually optional) IEs in the message. Here is an example:
```
In [1]: from pycrate_mobile.TS29274_GTPC import *                                                                                                         

In [2]: m = CreateBearerRequest()                                                                                                                         

In [3]: print(m.show())                                                                                                                                   
### CreateBearerRequest ###
 ### GTPCHdr ###
  <Version : 2>
  <P : 0>
  <T : 1>
  <MP : 0>
  <spare : 0>
  <Type : 95 (Create Bearer Request)>
  <Len : 8>
  <TEID : 0x00000000>
  <SeqNum : 0>
  <spare : 0>
     ### CreateBearerRequestIEs ###

In [4]: m[1].init_ies() # setting all mandatory IEs with dummy values                                                                                     

In [5]: print(m.show())                                                                                                                                   
### CreateBearerRequest ###
 ### GTPCHdr ###
  <Version : 2>
  <P : 0>
  <T : 1>
  <MP : 0>
  <spare : 0>
  <Type : 95 (Create Bearer Request)>
  <Len : 53>
  <TEID : 0x00000000>
  <SeqNum : 0>
  <spare : 0>
     ### CreateBearerRequestIEs ###
      ### LinkedEPSBearerID ###
       ### GTPCIEHdr ###
        <Type : 73 (EPS Bearer ID (EBI))>
        <Len : 1>
        <spare : 0>
        <Inst : 0>
           ### EBI ###
            <spare : 0>
            <Val : 5>
            <ext : 0x>
      ### BearerContext ###
       ### GTPCIEHdr ###
        <Type : 93>
        <Len : 36>
        <spare : 0>
        <Inst : 0>
           ### CreateBearerRequest_BearerContext ###
            ### EPSBearerID ###
             ### GTPCIEHdr ###
              <Type : 73 (EPS Bearer ID (EBI))>
              <Len : 1>
              <spare : 0>
              <Inst : 0>
                 ### EBI ###
                  <spare : 0>
                  <Val : 5>
                  <ext : 0x>
            ### BearerLevelQoS ###
             ### GTPCIEHdr ###
              <Type : 80 (Bearer Level Quality of Service (Bearer QoS))>
              <Len : 22>
[... 198 lines cut ...]
                  <Opcode : 1 (Create new TFT)>
                  <E : 0 (no parameters list)>
                  <NumPktFilters : 0>
                  ### PktFilters ###
      ### GTPCIE ###
       ### GTPCIEHdr ###
        <Type : 87 (Fully Qualified Tunnel Endpoint Identifier (F-TEID))>
        <Len : 9>
        <spare : 0>
        <Inst : 0>
           ### FTEID ###
            <V4 : 1>
            <V6 : 0>
            <IF : 0 (S1-U eNodeB GTP-U interface)>
            <TEID_GREKey : 0x99887766>
            <IPv4Addr : 0x0a0b0c0d>
            <ext : 0x>

In [37]: m.to_bytes()                                                                                                                                     
Out[37]: b'H_\x00B\x00\x00\x00\x00\x00\x00\x00\x00I\x00\x01\x00\x0c]\x00$\x00I\x00\x01\x00\x0cP\x00\x16\x00E\x07\x00\x00\x01\x86\xa0\x00\x00\x01\x86\xa0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00T\x00\x01\x00 W\x00\t\x00\x80\x99\x88wf\n\x0b\x0c\r'
```

All GTP-C messages from the few specifications related to this protocol are defined in the second part of the TS29274_GTPC.py file, and can be access through the `GTPCDispatcher` dict.

That's all I can provide to you. In case you come with more / better explanations or methods to work with GTP-C, please do not hesitate to write a new section for the pycrate wiki here: https://github.com/P1sec/pycrate/wiki/The-pycrate-wiki

akibsayyed, 2021-06-27:
Thank you @p1-bmu  for quick response

Let me test again with modified version

p1-bmu, 2022-02-01:
I just wrote this: https://github.com/P1sec/pycrate/wiki/Working-with-GTPC
Now closing this issue.

closed 2022-02-01 by p1-bmu