P1sec/pycrate #156: invalid undef count value, 63

coscar68, 2021-10-21:
I have tried to decode the string below (CAM message). It produces an 'Error:    invalid undef count value, 63' message.
This string is decoded without any issues by online decoders.

02020000D900B1E74059D824554CC4C2D79FFFFFFC2230D41E58622FC0000082B88A800FFD01FFF8807FE013C0400009FFFF7FFFD8CE00

p1-bmu, 2021-11-15:
I updated the compiler last week, for some specific cases, and introduce some of the new ETSI ITS ASN.1 module, including the CAM one. It seems now to work smoothly to decode your message:
```
In [1]: from pycrate_asn1dir.ITS_CAM_2 import *                                                                                                           

In [2]: m = CAM_PDU_Descriptions.CAM                                                                                                                      

In [3]: m.from_uper(unhexlify('02020000D900B1E74059D824554CC4C2D79FFFFFFC2230D41E58622FC0000082B88A800FFD01FFF8807FE013C0400009FFFF7FFFD8CE00'))          

In [4]: print(m.to_asn1())                                                                                                                                
{
  header {
    protocolVersion 2,
    messageID 2 -- cam --,
    stationID 55552
  },
  cam {
    generationDeltaTime 45543,
    camParameters {
      basicContainer {
        stationType 5 -- passengerCar --,
        referencePosition {
          latitude 421280170,
          longitude -86227780,
          positionConfidenceEllipse {
            semiMajorConfidence 4095 -- unavailable --,
            semiMinorConfidence 4095 -- unavailable --,
            semiMajorOrientation 3601 -- unavailable --
          },
          altitude {
            altitudeValue 0 -- referenceEllipsoidSurface --,
            altitudeConfidence unavailable
          }
        }
      },
      highFrequencyContainer basicVehicleContainerHighFrequency : {
        heading {
          headingValue 1570,
          headingConfidence 127 -- unavailable --
        },
        speed {
          speedValue 0 -- standstill --,
          speedConfidence 1 -- equalOrWithinOneCentimeterPerSec --
        },
        driveDirection unavailable,
        vehicleLength {
          vehicleLengthValue 44,
          vehicleLengthConfidenceIndication unavailable
        },
        vehicleWidth 18,
        longitudinalAcceleration {
          longitudinalAccelerationValue 0,
          longitudinalAccelerationConfidence 1 -- pointOneMeterPerSecSquared --
        },
        curvature {
          curvatureValue 1022,
          curvatureConfidence onePerMeter-0-00002
        },
        curvatureCalculationMode yawRateUsed,
        yawRate {
          yawRateValue 0 -- straight --,
          yawRateConfidence degSec-000-10
        },
        accelerationControl '0000000'B --  --,
        steeringWheelAngle {
          steeringWheelAngleValue 512 -- unavailable --,
          steeringWheelAngleConfidence 1 -- equalOrWithinOnePointFiveDegree --
        },
        lateralAcceleration {
          lateralAccelerationValue -2,
          lateralAccelerationConfidence 1 -- pointOneMeterPerSecSquared --
        }
      },
      lowFrequencyContainer basicVehicleContainerLowFrequency : {
        vehicleRole default,
        exteriorLights '00'H --  --,
        pathHistory {
          {
            pathPosition {
              deltaLatitude 0,
              deltaLongitude 0,
              deltaAltitude 0
            }
          }
        }
      }
    }
  }
}
```

coscar68, 2021-11-15:
Hi, I have downloaded current version, and run it again but I get the same error. I do not know if I am missing something.

p1-bmu, 2021-12-01:
You certainly have still an old install somewhere.

closed 2022-03-28 by p1-bmu