P1sec/pycrate #159: ASN.1 UTF8String encode/decode

palyamate, 2021-11-05:
I used this scheme:
```
PERSON

DEFINITIONS AUTOMATIC TAGS ::=

BEGIN

Person ::= SEQUENCE {
    age               INTEGER(0..99),
    firstname         UTF8String,
    lastname          UTF8String
}

END
```
I encoded this on [this](https://asn1.io/asn1playground/):
```
{
    "age": 72,
    "firstname": "JSKHFDGhgha",
    "lastname": "HSjkhasjfhkaj"
}
```
and the UPER encoded version:
`0148044A 6F686E03 446F65`
I compiled this with _pycrate_asn1compile.py_ and used the `.from_uper` function of  the Person.
I got this error:
```
Traceback (most recent call last):
  File "name_decode.py", line 4, in <module>
    person.PERSON.Person.from_uper(bytes.fromhex('0148044A 6F686E03 446F65'))
  File "/home/palya/work/pycrate-investigation/lib/python3.8/site-packages/pycrate_asn1rt/asnobj.py", line 1292, in from_uper
    self._from_per(char)
  File "/home/palya/work/pycrate-investigation/lib/python3.8/site-packages/pycrate_asn1rt/asnobj_construct.py", line 949, in _from_per
    Comp._from_per(char)
  File "/home/palya/work/pycrate-investigation/lib/python3.8/site-packages/pycrate_asn1rt/asnobj_str.py", line 2264, in _from_per
    self.__from_per_szunconst(char)
  File "/home/palya/work/pycrate-investigation/lib/python3.8/site-packages/pycrate_asn1rt/asnobj_str.py", line 2326, in __from_per_szunconst
    self.__from_per_charstr(char, ldet)
  File "/home/palya/work/pycrate-investigation/lib/python3.8/site-packages/pycrate_asn1rt/asnobj_str.py", line 2332, in __from_per_charstr
    val = char.get_bytes(8*ldet)
  File "/home/palya/work/pycrate-investigation/lib/python3.8/site-packages/pycrate_core/charpy.py", line 375, in get_bytes
    raise(CharpyErr('bitlen overflow: {0}, max {1}'\
pycrate_core.charpy.CharpyErr: bitlen overflow: 73744, max 65
```
The standard is [here](https://www.itu.int/ITU-T/studygroups/com17/languages/X.691-0207.pdf). 
The 10.9 says something about the string lenght encoding.

p1-bmu, 2021-11-08:
Thanks for your feedback.
Your encoding `0148044A 6F686E03 446F65` does not make really sense, as it looks shorter than the firstname / lastname strings, which are unconstrained UTF8, meaning each character is encoded on at least 1 byte. 
You can also see with the asn1playground that decoding this buffer does not work: 
```
D0081S: End of input reached before message was fully decoded; check field 'firstname' of PDU #1.
value: D0081S: End of input reached before message was fully decoded; check field 'firstname' of PDU #1.
S0014E: Printing of PER details failed with the return code '4'.

The output may be truncated.
```

palyamate, 2021-11-08:
Oh, sorry, I copied the wrong structure and bytes...
This is the encoded version:
`48044A6F 686E0344 6F65`
And this is the correct JSON:
```
{
  "age": 72
  "firstname": "John",
  "lastname": "Doe"
}
```

p1-bmu, 2021-11-08:
```
Encoding to the file 'data.uper' using PER UNALIGNED encoding rule...
Person SEQUENCE [fieldcount (not encoded) = 3]
  age INTEGER [length (not encoded) = 0.7]
    72
  firstname UTF8String [length = 4.0]
    0x4a6f686e
  lastname UTF8String [length = 3.0]
    0x446f65
Total encoded length = 9.7
Encoded successfully in 10 bytes:
900894DE D0DC0688 DECA
```
You are failing somewhere with the value you want to encode or the buffer you have encoded. Please verify you data before submitting further information / issue.

palyamate, 2021-11-08:
```
PERSON

DEFINITIONS AUTOMATIC TAGS ::=

BEGIN

Person ::= SEQUENCE {
    age               INTEGER(0..99),
    firstname         UTF8String,
    lastname          UTF8String
}

END
```
```
{
  "age": 72,
  "firstname": "John",
  "lastname": "Doe"
}
```
`900894DE D0DC0688 DECA`

closed 2021-11-08 by palyamate