P1sec/pycrate #162: Question about tagging in CHARACTER STRING

russhousley, 2021-11-26:


russhousley, 2021-11-26:
The tag for string-value is set to 1.

~~~
    _CHARACTER_STRING_string_value = OCT_STR(name='string-value', mode=MODE_TYPE, tag=(1, TAG_CONTEXT_SPEC, TAG_EXPLICIT))
~~~

I expected data-value-descriptor to have a tag of 1, and string-value to have a tag of 2.

I realize that the constraints mean that data-value-descriptor will never be present, but I think that AUTOMATIC TAGS still assigns a tag number to it.

~~~
( WITH COMPONENTS { ... , data-value-descriptor ABSENT } )
~~~

p1-bmu, 2021-11-29:
CHARACTER STRING is a predefined object, built here: https://github.com/P1sec/pycrate/blob/26c1a0a6bd26f9b0ab8f22e6efc619cbda3642b3/pycrate_asn1c/asnproc.py#L531
Components are explicitly tagged, so there is no AUTOMATIC TAGS implied.

What is your exact question ?

russhousley, 2021-11-29:
In the ASN.1 specification, the syntax for EmbeddedPDV, EXTERNAL, and CHARACTER STRING are provided, and the text says that AUTOMATIC TAGS are used.  I believe that the data-value-descriptor gets a tag of 1 (but it is never present) and that string-value should be given a tag of 2.

p1-bmu, 2021-12-01:
You are correct: I was missing the `data-value-descriptor` component in the CHARACTER STRING definition. I just fixed it in master.
Thank you.

closed 2021-12-01 by russhousley