P1sec/pycrate #166: argument of type 'NoneType' is not iterable

igrag, 2021-12-13:
Hello,
I am trying to compile the IS TS 103 301 from ETSI (https://forge.etsi.org/rep/ITS/asn1/is_ts103301). These are the asn files for the IVI messages. I am getting the folllowing error:

**__Exception has occurred: TypeError
argument of type 'NoneType' is not iterable
  File "/home/grag/work/opt-unex-software/pycrate/pycrate_asn1c/asnobj.py", line 4337, in __parse_const_withcomps_comp
    if ident not in cont:
  File "/home/grag/work/opt-unex-software/pycrate/pycrate_asn1c/asnobj.py", line 4284, in _parse_const_withcomps
    self.__parse_const_withcomps_comp(rc, const_comp)
  File "/home/grag/work/opt-unex-software/pycrate/pycrate_asn1c/asnobj.py", line 3976, in _parse_const
    self._parse_const_withcomps(const)
  File "/home/grag/work/opt-unex-software/pycrate/pycrate_asn1c/asnobj.py", line 1716, in parse_def
    rest = self._parse_const(text)
  File "/home/grag/work/opt-unex-software/pycrate/pycrate_asn1c/asnobj.py", line 3418, in __parse_cont_comps
    comp_t = Comp.parse_def(comp_t)
  File "/home/grag/work/opt-unex-software/pycrate/pycrate_asn1c/asnobj.py", line 3387, in _parse_cont_seq
    self.__parse_cont_comps(text_comps, self._type, False)
  File "/home/grag/work/opt-unex-software/pycrate/pycrate_asn1c/asnobj.py", line 2269, in _parse_cont
    text = getattr(self, self._PARSE_CONT_DISPATCH[self._type])(text)
  File "/home/grag/work/opt-unex-software/pycrate/pycrate_asn1c/asnobj.py", line 1705, in parse_def
    text = self._parse_cont(text)
  File "/home/grag/work/opt-unex-software/pycrate/pycrate_asn1c/asn_obj.py", line 3198, in _parse_cont_seqof
    text = Comp.parse_def(text)
  File "/home/grag/work/opt-unex-software/pycrate/pycrate_asn1c/asnobj.py", line 2269, in _parse_cont
    text = getattr(self, self._PARSE_CONT_DISPATCH[self._type])(text)
  File "/home/grag/work/opt-unex-software/pycrate/pycrate_asn1c/asnobj.py", line 1705, in parse_def
    text = self._parse_cont(text)
  File "/home/grag/work/opt-unex-software/pycrate/pycrate_asn1c/asnproc.py", line 879, in asnobj_compile
    text = Obj.parse_def(Obj._text_def)
  File "/home/grag/work/opt-unex-software/pycrate/pycrate_asn1c/asnproc.py", line 945, in compile_modules
    ObjNew = asnobj_compile(Obj)
  File "/home/grag/work/opt-unex-software/pycrate/pycrate_asn1c/asnproc.py", line 257, in compile_text
    compile_modules(remain)
  File "/home/grag/work/opt-unex-software/pycrate/tools/pycrate_asn1compile.py", line 219, in main
    compile_text(txt, **ckw)
  File "/home/grag/work/opt-unex-software/pycrate/tools/pycrate_asn1compile.py", line 267, in <module>
    sys.exit(main())_**

I think that the error occurs in the following statement in one of the asn files:

**GetInstanceRs{Container}::= SEQUENCE (SIZE (0..127,...)) OF SEQUENCE {
attributeId INTEGER(0..127,...),
-- number of instances retrieved
attributeValues Container (WITH COMPONENTS {octetstring PRESENT})
-- The octetstring shall contain the contatenation of
-- the unaligned PER encodings of the values of the
-- instances, with each encoding padded to an integral
-- of octets as specified for a top-level type in
-- ISO/IEC 8825-2
}**

If I comment out this then the compilation completes successfully.
Is this a limitation of pycrate or a problem in the asn file?

Best Regards,
Giannis

p1-bmu, 2021-12-14:
Thanks for reporting. This may correspond to an issue in the compiler, or in the spec.

Regarding this ETSI IS specification, it relies on an ISO TS 19321 standard which is not free, and not part of the ETSI git repository. Hopefully, one can access the ASN.1 modules here: https://standards.iso.org/iso/ts/19321/. But then, this last one requires the module `CITSdataDictionary1 {iso(1) standard(0) cits-applMgmt (17419) dataDictionary (1) version1 (1)}`, which I cannot find on the ISO website (looking at https://standards.iso.org/iso/ts/17419/).

So, unless you provide the full ASN.1 specification for the compilation to work, I won't be able to reproduce.

igrag, 2021-12-14:
Hello,
Thank you for the answer and your help.
Attached are the asn files that I have used.

Best Regards
Giannis Gkragkopoulos

p1-bmu, 2021-12-14:
The attached set of asn1 files does not have any `CITSdataDictionary1` module, unfortunately.

igrag, 2021-12-14:
I am sorry.  I uploaded wrong files
Please use these.
Thank you
[ETSI_ITS_IS_TS103301_ORIGINAL_NEW.zip](https://github.com/P1sec/pycrate/files/7711803/ETSI_ITS_IS_TS103301_ORIGINAL_NEW.zip)
.

p1-bmu, 2021-12-14:
This one is also missing the `CITSdataDictionary1` module.

p1-bmu, 2021-12-15:
I fixed the compiler to bypass the processing of WITH COMPONENTS constraints in case of parameterization, which avoids the error. On the other side, we are still missing `CITSdataDictionary1`, as used in the `IVI.asn` file.

igrag, 2021-12-15:
Thank you. 
I have no problem to include it in the library with the other standards. I have't tested anything yet, so I don't know if it actually works. I took the files from here: https://forge.etsi.org/rep/ITS/asn1. Although we are members of ETSI I didn't use my credentials so I think that all the ASNs are open.

Thanks again for your time and effort.

p1-bmu, 2021-12-15:
OK, done. Now available as: `from pycrate_asn1dir import ITS_IS`.

closed 2021-12-15 by p1-bmu