P1sec/pycrate #167: Need a sample MSISDN initialization example

sohel1119, 2021-12-28:
I am perform some activity regarding GTP-V2-C and not getting actual output of MSISDN

p1-bmu, 2021-12-28:
https://github.com/P1sec/pycrate#filling-an-issue

sohel1119, 2021-12-28:
Python 3.8.10 (default, Nov 26 2021, 20:14:08) 

In [1]: from pycrate_mobile.TS29274_GTPC import *

In [2]: m = CreateSessionRequest()
   ...: m[1].init_ies()
   ...: m[1].add_ie(76,1)     

In [3]: print(m[1][4][1].show())
        <MSISDN : >
        
In [4]: m[1][4][1].set_val({'MSISDN ': 1234567890})
   ...: print(m[1][4][1].show())
        <MSISDN : >

p1-bmu, 2021-12-29:
```
In [20]: m[1][4][1].set_val('1234567890')                                                                                                                 

In [21]: m[1][4][1]                                                                                                                                       
Out[21]: <MSISDN : 1234567890>
```

closed 2022-01-24 by p1-bmu