P1sec/pycrate #168: Feature Group Indicators LTE UE Capability

ritgail, 2022-01-04:
Hi Developers,

I was trying to decode UE capability information on LTE UL_DCCH message. I used the below codes:
# m = lte_ue_capability(unhexlify('ddb8050601382081d40b0d1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffbfe7ff5f441401190f04194e952547f027f380000000a9df00000002ffc00c00c0090028210c00c1006084104218210c10060803040182104100608030421801821041006084104208210c00c100608030401821041006080304208210c00c1006084108010c2080108010c2080108000c2084108000c208410401821041006084108000c2080108000c2080108000c2004108010c20003080104200030801042004308000c20801080104200030820042004104018210420003082004200030820042000308010420003080104200030801042084108000c108208030821042080108000c2080108000c1082104008400061040084000610021840086100218200c10061000184100210020840006104008400821002184100210021841002180010c300021860004308000c2004108000c2004108000c2000308000c2000304018018018018018200c2004308000969f5518c63f8180000820738002081840008206200020018e00080074000208196000800e000020a004103000012416708180000820b3841c0000400708001050020828410700001041c400040400820a2040c00004105a020600002082d010700001001c700040400820a3840c000041079c001059c2081ce000920b3841c200040059c20610002082ce10710001001670838e000800e100020a10410308001241c400040420820a2040c20004105a020610002082d010708001001c700040420820a3840c20004105a020e38002003d000082d01040e800049059c24600002082ce12700001002c20004040082801042c20004140080801002420818000082801042420838000080801002c20004042080801042c20004142082801002420808410300001042420828410700001001c200040400920e100020a00480508206000024828410700001202670818800080801042670838800080801002670808810300001042670828810300001002c40004042080801042c40004042082801002c400040400828010424408184000808010424408184000828010024408084103000010424408084107000010024408180000828010426808184000808010426808384000808010026808184000828010026808084103000010426808284103000010026808084107000010016809180000820b4049c000040099c20638002020041099c20e38002020040099c2023840c000041099c20a3840c0000400b1c001010820200410b1c001010820a0040091c20610002020041091c206100020a0040091c2021040c000041091c2021041c000040099c20b3841c200040099c2033840c200041059c24e10002002ce1230800104267082ce10710001002670838800080841002670818800080841042670828810308001002670808810308001041670938800080133841c70004042080133840c70004042082133841470818400080133840470818400082134040c20004142082163800202104142082123840c200041420820e38002021049051c20610002484ce1230800101002084ce1270800101002004ce1230800105002004ce1210820600002084ce1250820600002004ce1210820e000020058400090801050020858400092801010020048412300001050020848412700001010020078800080841010020a00410d1020610002020041400821a2040420818000082801042c40004042090801042c40004042092801002440818400090801042440818400092801002440808412300001042c700040420908010424708184000908010424708084123000010416709984000820600082481ce020920610082481d0020920a3840c2010490b08041210020a0041091c20610082420041ff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4fff1bf93ffc6fe4ffef9d800011ff008721e1e1e1e1e1e2f178787878787878ba4fc9f87878fe3f87878fc0000878743c3c3c380023f8fe3f8fe3f8fc10f0f0f0f0e08787000000878747e1f82009001f32bcd0200a46ffff34a5de38074002c1004c0c0029249000020005104818004104200'))
# with open('uecapability.txt', 'w') as f:
#     print(request.to_asn1(), file=f)

def lte_ue_capability(payload):
    request = RRCLTE.EUTRA_RRC_Definitions.UE_EUTRA_Capability
    request.from_uper(payload)
    jsonObject = json.loads(request.to_json())
    return jsonObject

I am able to get the result, but I need feature group indicator values. Currently, it is showing in Hex.

Is it already implemented to decode? 
Thanks for your support.
[uecapability.txt](https://github.com/P1sec/pycrate/files/7806215/uecapability.txt)

p1-bmu, 2022-01-04:
`featureGroupInd*` are all defined as `BIT STRING (SIZE (32))` in the LTE RRC ASN.1 specification. Pycrate does not provide additional decoding for those fields.
You need to check annexes B.1 and C.1 of the 3GPP TS 36.331, which describe how those fields are encoded (what seems fairly complex at 1st sight...).

ritgail, 2022-01-05:
I can try to populate these fields similar to wireshark.

Could you guide me how to do that with the BIT STRING. I can see it is now like:
_UE_EUTRA_Capability_featureGroupIndicators = BIT_STR(name=u'featureGroupIndicators', mode=MODE_TYPE, tag=(6, TAG_CONTEXT_SPEC, TAG_IMPLICIT), opt=True)
    _UE_EUTRA_Capability_featureGroupIndicators._const_sz = ASN1Set(rv=[32], rr=[], ev=None, er=[])

[image]

p1-bmu, 2022-01-05:
Ideally, you would have to change the ASN.1 definitions of the different objects `featureGroupInd*` in the file EUTRA-RRC-Definitions.asn, to define named bits (see e.g. https://www.oss.com/asn1/resources/asn1-made-simple/types.html). You would have then to recompile the LTERRC ASN.1 spec.
Finally, you should be able to access the named bits after decoding a buffer through one of the following methods: 
- https://github.com/P1sec/pycrate/blob/4efc57b6750c65789d2d0dadfc0f30f31febf36d/pycrate_asn1rt/asnobj_str.py#L161
- https://github.com/P1sec/pycrate/blob/013b3fccbb3025ea73a900c10e3c946006b4901d/pycrate_asn1rt/asnobj.py#L1225

closed 2022-03-28 by p1-bmu