P1sec/pycrate #180: Decoding with output in json format

dbressan2, 2022-02-15:
Hi 

I am trying to get decoding results in JSON format. I have tried two ways, none of the two being completely satisfying. The issues are with the ordering of decoding results and decoding of BIT STRING. 
Decoding with to_asn1() method gives nice results, which I would like to translate into JSON 

method 1: 
using json.dump with the dictionary returned by invoking dcch() with dcch = NR_RRC_Definitions.UL_DCCH_Message() 
Works mostly well, but instead of LocationSource-r13: 'E'H -- a-gnss | wlan | bt -- 
we get  
 "LocationSource-r13",
                                        [
                                            14,
                                            4
                                        ]
Hence value, length of BIT STRING 

method 2: 
dcch.to_json() 
Issue is that dictionary order is completely messed up with bits of message appearing out of order. 
However BIT STRING decoding is better 
         "locationSource-r16": {
          "LocationSource-r13": {
           "length": 4,
           "value": "e0"
          }
         }, 

Full code: 

class MyEncoder(json.JSONEncoder):
    def default(self, obj):
        # if isinstance(obj, np.ndarray):
        #     return obj.tolist()
        if isinstance(obj, bytes):
            return obj.hex()
        return json.JSONEncoder.default(self, obj)

message = """
00 c2 00 04 63 df 24 2b b0 10 7e 48 57 41 15 b7
5c d5 34 2c 0e bd c2 a8 fd 73 64 f1 ba c5 e7 a1
75 8d 53 43 ea d9 22 89 d5 30 44 08 91 21 1d 01
d6 1a c0 07 80 6b 26 4c 19 30 70 c1 8b 46 cd 19
da 75 80 04 a1 83 f6 16 6b 1b 38 0d 20 00 9F 00
""".replace(" ", "").replace("\n", "")

dcch = NR5G_Dec_2021.NR_RRC_Definitions.UL_DCCH_Message 
loc_source = LPP_Dec_2021.LPP_PDU_Definitions.LocationSource_r13

_cont_locationTimestamp_r16 = CHOICE(name='DisplacementTimeStamp-r15', mode=MODE_TYPE, 
                                     typeref=ASN1RefType(('LPP-PDU-Definitions', 'DisplacementTimeStamp-r15')))

_cont_locationCoordinate_r16 = CHOICE(name='LocationCoordinates', mode=MODE_TYPE, 
                                     typeref=ASN1RefType(('LPP-PDU-Definitions', 'LocationCoordinates')))

_cont_locationSource_r16 = BIT_STR(name='LocationSource-r13', mode=MODE_TYPE, 
                                     typeref=ASN1RefType(('LPP-PDU-Definitions', 'LocationSource-r13')))

loc_info = ['message', 'c1', 'measurementReport', 'criticalExtensions', 'measurementReport', 'measResults', 
            'locationInfo-r16', 'commonLocationInfo-r16']
dcch.get_at(loc_info + ['locationTimestamp-r16'])._const_cont = _cont_locationTimestamp_r16
dcch.get_at(loc_info + ['locationTimestamp-r16'])._const_cont_enc = None 
dcch.get_at(loc_info + ['locationCoordinate-r16'])._const_cont = _cont_locationCoordinate_r16
dcch.get_at(loc_info + ['locationCoordinate-r16'])._const_cont_enc = None 
dcch.get_at(loc_info + ['locationSource-r16'])._const_cont = _cont_locationSource_r16
dcch.get_at(loc_info + ['locationSource-r16'])._const_cont_enc = None 
from pycrate_asn1rt.init import init_modules 
extensions = [_cont_locationTimestamp_r16, _cont_locationCoordinate_r16, _cont_locationSource_r16]
NR5G_Dec_2021.NR_RRC_Definitions._all_.extend(extensions)
init_modules(NR5G_Dec_2021.NR_RRC_Definitions) 

dcch.from_uper(bytes.fromhex(message))

print (dcch.to_asn1())

# solution 1 
json_object = json.dumps(dcch(), cls=MyEncoder, indent = 4)
print (json_object)

# solution 2
print(dcch.to_json())

p1-bmu, 2022-02-16:
I don't understand what you exactly want to do, but from what you are indicating, maybe you can call `.convert_named_val()` on the `dcch` object, before collecting its value, and then serializing the value to JSON ?
https://github.com/P1sec/pycrate/blob/5f5f151358a03641ce65374d014851cc8271b164/pycrate_asn1rt/asnobj.py#L1225

dbressan2, 2022-02-17:
> WARNING: This email originated from outside of Qualcomm. Please be wary of any links or attachments, and do not enable macros.

Hi

Thanks, yes using convert_named_val method and then converting Python set to list prior to serializing to JSON addresses my problem

Thank you very much for your help

Best regards

Dominique

https://github.com/P1sec/pycrate/blob/5f5f151358a03641ce65374d014851cc8271b164/pycrate_asn1rt/asnobj.py#L1225

closed 2022-02-17 by p1-bmu