P1sec/pycrate #184: i want to add Cont ,NCPOpt ,ProtConfigElt in PCO in GTP-C Create sesstion massage

sohel1119, 2022-03-28:
# Python version 3.8.10
# PyCrate version 0.5.3

from pycrate_mobile.TS29274_GTPC import *
m = CreateSessionRequest()
m[1].add_ie(78,0)
print(m[1].show())
'''output-
    ### CreateSessionRequestIEs ###
     ### PCO ###
      ### GTPCIEHdr ###
       <Type : 78 (Protocol Configuration Options (PCO))>
       <Len : 1>
       <spare : 0>
       <Inst : 0>
          ### PCO ###
           <Ext : 1>
           <spare : 0>
           <Prot : 0 (PPP with IP PDP)>
           ### Config ###'''

m[1][0][1][3].set_val([{'ID':13}])
print(m[1].show())
'''output-
    ### CreateSessionRequestIEs ###
     ### PCO ###
      ### GTPCIEHdr ###
       <Type : 78 (Protocol Configuration Options (PCO))>
       <Len : 4>
       <spare : 0>
       <Inst : 0>
          ### PCO ###
           <Ext : 1>
           <spare : 0>
           <Prot : 0 (PPP with IP PDP)>
           ### Config ###
            ### ProtConfigElt ###
             <ID : 13 (DNS server IPv4 address request)>
             <Len : 0>
             <Cont : 0x>
'''
# im not able to add Cont ,NCPOpt ,ProtConfigElt
# following output i want by using Cont ,NCPOpt ,ProtConfigElt will you help me for that
'''
### PCO ###
     ### GTPCIEHdr ###
      <Type : 78 (Protocol Configuration Options (PCO))>
      <Len : 45>
      <spare : 0>
      <Inst : 0>
         ### PCO ###
          <Ext : 1>
          <spare : 0>
          <Prot : 0 (PPP with IP PDP)>
          ### Config ###
           ### ProtConfigElt ###
            <ID : 32801 (IPCP)>
            <Len : 16>
            ### Cont ###
             <Code : 1 (Configure-Request)>
             <Id : 0>
             <Len : 16>
             ### Data ###
              ### NCPOpt ###
               <Type : 129 (Primary DNS Server Address)>
               <Len : 6>
               <Data : 0x00000000>
              ### NCPOpt ###
               <Type : 131 (Secondary DNS Server Address)>
               <Len : 6>
               <Data : 0x00000000>
           ### ProtConfigElt ###
            <ID : 13 (DNS server IPv4 address request)>
            <Len : 0>
            <Cont : 0x>
           ### ProtConfigElt ###
            <ID : 10 (IP address allocation via NAS signalling)>
            <Len : 0>
            <Cont : 0x>
           ### ProtConfigElt ###
            <ID : 5 (Selected Bearer Control Mode)>
            <Len : 0>
            <Cont : 0x>

 '''In PCO (IE type 78)  i want  to add Cont ,NCPOpt ,ProtConfigElt as shown in above output but im not able to add it. 
can you guide me ?

mitshell, 2022-03-28:
Could you please, at least, do some efforts to make your issue readable ?

sohel1119, 2022-03-28:
Ok, As you can see I have attached the screen shot, I want to create exact same Output.
[image]