P1sec/pycrate #185: Self._ext 'NoneType' object is not iterable

samuel-rangel, 2022-04-01:
I'm trying to encode in ber some telco data with my own defined asn1 structure.
The compiler doesn't throw any error, I can use the created class and set value to my object with set_value() method, I can even view the data in the object with the to_asn1() method, but when I try to encode it with the to_ber() method it throws me an error.
`File "C:\Users\samue\AppData\Local\Programs\Python\Python310\lib\site-packages\pycrate-0.5.4-py3.10.egg\pycrate_asn1rt\asnobj_construct.py", line 2587, in _encode_ber_cont
    for ident in self._ext:
TypeError: 'NoneType' object is not iterable`
Clearly ._ext field is None, but for what I read in the docs it is optional, and actually I've been able to encode other structures that are less complex that also had ._ext field None.
My structure definition and some dummy data sample is attached
[VoLTE.zip](https://github.com/P1sec/pycrate/files/8400167/VoLTE.zip)

p1-bmu, 2022-04-04:
`SET` is rarely used in ASN.1, so there may be a missing check lying in the runtime for years.
In order to fix this, I need however to understand exactly what happens. I could compile your `VoLTE.asn` file, could you please tell me how to reproduce your issue exactly ?

samuel-rangel, 2022-04-04:
Thanks for replying!
Sure, here is a snippet that I'm using for testing
[test_VoLTE_encode.zip](https://github.com/P1sec/pycrate/files/8411066/test_VoLTE_encode.zip)

closed 2022-04-05 by samuel-rangel