P1sec/pycrate #186: customizing to_asn1 output

dbressan2, 2022-04-04:
Hi 

I would like to know the best way to change the look of to_asn1 output for various objects. 
I tried to modify the code in pycrate_asn1/asnobj.py. 
Let us take an example 

Say I want to print the TIME_UTC output as YYMMDDHHMMSSUTC instead of YYMMDDHHMMSSZ. 

Is it enough to modify TIME_UTC code ? Do we need to re-build package afterwards ? 

class TIME_UTC(_Time):
    __doc__ = """
    ASN.1 basic type UTCTime object
    
    single value: Python 7-tuple of int (AA, MM, DD, HH, MM, [SS,] Z),
        SS is optional, hence 6th element can be None
        Z corresponds to the UTC decay
    
    %s
    """ % ASN1Obj_docstring
    
    TYPE  = TYPE_TIME_UTC
    TAG   = 23
    _type = TYPE
    
    def _to_asn1(self, val):
        # to be applied to an internal single value `val' to get 
        # an ASN.1 compliant value
        ret = '%.2i%.2i%.2i%.2i%.2i' % val[:5]
        if val[5] is not None:
            ret += '%.2i' % val[5]
        if val[6] == 0:
            ret += 'Z'
        elif val[6] > 0:
            ret += '+%.4i' % val[6]
        else:
            ret += '%.4i' % val[6]
        return ret

Thank you in advance

p1-bmu, 2022-04-04:
As soon as you patch a file of the library, the change will be effective as soon as you load it (e.g. in case you use only a local version from your $HOME, or you installed with `setup.py install develop`): no need for re-packaging. This is one of the advantage of Python.

Regarding the modification of the generic `to_asn1()` in `asnobj.py`, or the various `_to_asn1()` methods, this will change the formatting of the output for sure. On the other side, `to_asn1()` is made to produce ASN.1-compliant syntax and formatting (its output could be ingested by some other ASN.1 tools), it may not be the case anymore if you change those methods.

dbressan2, 2022-04-07:
> WARNING: This email originated from outside of Qualcomm. Please be wary of any links or attachments, and do not enable macros.

Thank you for your quick reply

I am trying to modify the code in pycrate_asn1rt/asnobjcontruct.py

The goal is systematically number items within a list.

    def _to_asn1(self):
        if not self._val:
            # empty dict
            return '{ }'
        else:
            val = []
            # root and ext part in 1 shot
            # WNG: we are not ordering SET root components in canonical order
            for ident in self._cont:
                if ident in self._val:
                    print ("ident = {0}".format(ident))
                    _par = self._cont[ident]._parent
                    self._cont[ident]._parent = self
                    self._cont[ident]._val = self._val[ident]
                    print ('  %s %s,\n' % (ident, self._cont[ident]._to_asn1().replace('\n', '\n  ')))
                    val.append('  %s %s,\n' % (ident, self._cont[ident]._to_asn1().replace('\n', '\n  ')))
                    self._cont[ident]._parent = _par
            if val:
                val[-1] = val[-1][:-2]
            return '{\n' + ''.join(val) + '\n}'

If we take the example of an NR RRC DL-DCCH message

Message =
00 0c 02 48 20 08 44 02 92 48 00 30 00 0a 24 00
18 00 02 12 42 94 0a 10 84 1c 6b 80 80 9c 08 00
00 3d 06 24 30 54 06 e0 40 00 01 e8 31 26 82 a5
05 28 14 21 08 38 d7 01 01 38 10 00 00 7a 0c 48
60 a8 0d c0 80 00 03 d0 62 4d 05 40 00

In blue, the decoding as it is right now after invoking to_asn1 method, in red what I would like to add in a systematic way, without having to delve into each message with get_at() or get_val_at() methods.
Is there a simple modification of _to_asn1 that may achieve the item numbering ?

{
  message c1 : rrcReconfiguration : {
    rrc-TransactionIdentifier 0,
    criticalExtensions rrcReconfiguration : {
      nonCriticalExtension {
        masterCellGroup CellGroupConfig: {
          cellGroupId 0,
          spCellConfig {
            spCellConfigDedicated {
              downlinkBWP-ToAddModList {
                {
                  Item 0
                  bwp-Id 1,
                  bwp-Dedicated {
                    pdsch-Config setup : {
                      resourceAllocation resourceAllocationType0,
                      rbg-Size config1,
                      mcs-Table qam256,
                      maxNrofCodeWordsScheduledByDCI n1,
                      prb-BundlingType staticBundling : { }
                    }
                  }
                },
                {
                  Item 1
                  bwp-Id 2,
                  bwp-Dedicated {
                    pdsch-Config setup : {
                      resourceAllocation resourceAllocationType0,
                      rbg-Size config1,
                      mcs-Table qam256,
[... 150 lines cut ...]
                              periodicityAndOffset-p sl40 : 13
                            },
                            sequenceId 21
                          }
                        }
                      }
                    }
                  }
                }
              },
              tag-Id 0,
              servingCellMO 1
            }
          }
        }
      }
    }
  }
}

Thank you in advance

Best regards

Dominique

p1-bmu, 2022-04-11:
I don't really understand what / how you want to number items. I see no colour in your provided description, sorry.

dbressan2, 2022-04-11:
Hi

What I wish to add is highlighted below with the mark “<=== to be added”

{
  message c1 : rrcReconfiguration : {
    rrc-TransactionIdentifier 0,
    criticalExtensions rrcReconfiguration : {
      nonCriticalExtension {
        masterCellGroup CellGroupConfig: {
          cellGroupId 0,
          spCellConfig {
            spCellConfigDedicated {
              downlinkBWP-ToAddModList {
                {
                  Item 0 <=== to be added
                  bwp-Id 1,
                  bwp-Dedicated {
                    pdsch-Config setup : {
                      resourceAllocation resourceAllocationType0,
                      rbg-Size config1,
                      mcs-Table qam256,
                      maxNrofCodeWordsScheduledByDCI n1,
                      prb-BundlingType staticBundling : { }
                    }
                  }
                },
                {
                  Item 1 <=== to be added
                  bwp-Id 2,
                  bwp-Dedicated {
                    pdsch-Config setup : {
                      resourceAllocation resourceAllocationType0,
                      rbg-Size config1,
                      mcs-Table qam256,
                      maxNrofCodeWordsScheduledByDCI n1,
                      prb-BundlingType staticBundling : { }
                    }
                  }
                }
              },
              uplinkConfig {
                uplinkBWP-ToAddModList {
                  {
                    Item 0 <=== to be added
                    bwp-Id 1,
                    bwp-Dedicated {
                      srs-Config setup : {
                        srs-ResourceSetToAddModList {
                          Item 0 <=== to be added
                          {
                            srs-ResourceSetId 2,
                            srs-ResourceIdList {
                              2,
                              3
                            },
                            resourceType periodic : { },
                            usage antennaSwitching,
                            p0 -110
                          }
                        },
                        srs-ResourceToAddModList {
                          {
                            Item 0 <=== to be added
                            srs-ResourceId 2,
                            nrofSRS-Ports ports2,
                            transmissionComb n4 : {
                              combOffset-n4 2,
                              cyclicShift-n4 0
                            },
[... 115 lines cut ...]
                              periodicityAndOffset-p sl40 : 13
                            },
                            sequenceId 21
                          }
                        }
                      }
                    }
                  }
                }
              },
              tag-Id 0,
              servingCellMO 1
            }
          }
        }
      }
    }
  }
}

Thank you very much for your help

Best regards

Dominique

dbressan2, 2022-10-11:
Thank you so much, this is exactly what I was looking for.

Yes, I understand the limitation about the ASN.1 syntax not being re-encodable. However this is not blocking to me

Best regards

Dominique

You can modify the _to_asn1() method of the _CONSTRUCT_OF class here: https://github.com/P1sec/pycrate/blob/1a6d4d6ab8b61e0ef4ea9314ac07facfdd10f785/pycrate_asn1rt/asnobj_construct.py#L2686
To have a pattern like this instead:

            for i, v in enumerate(self._val):

                self._cont._val = v

                val.append('  -- _item_ %i --\n  %s,\n' % (i, self._cont._to_asn1().replace('\n', '\n  ')))

I guess such generated ASN.1 syntax won't be re-encodable with pycrate, as its runtime does not strip ASN.1 comments out.

closed 2022-12-08 by p1-bmu