P1sec/pycrate #188: Creating samples messages of ASN1 objects

karim140995, 2022-04-13:
Im trying to create sample messages for all of the RRC NR existing objects without having the binary format of the object. I checked the docs and the only way possible is to incrementally set the values of the objects with the set_val_at method. This is not very efficient given the complexity of the specification, is there a way around this ?

p1-bmu, 2022-04-19:
You have different methods to set appropriate values into ASN.1 object : `set_val()`, `set_val_at()`, as explained in the [wiki](https://github.com/P1sec/pycrate/wiki/Using-the-pycrate-asn1-runtime). There is nothing more provided.
Each ASN.1 protocol has its own specificity, which may require dedicated routines for easing the setting of values. This is up to you to build your own for the RRC NR one, I believe.

p1-bmu, 2022-04-26:
It seems maybe a similar effort was intended by another user, without feedback unfortunately: https://github.com/P1sec/pycrate/issues/104

karim140995, 2022-05-05:
For anyone looking for a time effective solution to this problem, OSS Noklava ASN1Studio offers the possibility to generate customized ASN messages in an automated way.

closed 2023-09-08 by p1-bmu