P1sec/pycrate #189: Error handling NAS Attch Request elements

abouillot, 2022-05-05:
Hello,

I use pycrate to analysis cellular phone logs.

I get the hex input from the log file and parse with parse_NAS_MO method. The message is an Attach request

The parsing goes fine with the various messages from different phones. When using the show function the content of the NAS message is properly displayed in all cases so far, with the proper items.

However, when trying specific items in the message, from one model, I systematically get an error with one message complaining the UENetCap is missing, while it is visible when looking into the NAS message.

Here is the code to show the behavior, with the message from one model correctly handled and the other failing:

```
    # ok
    msg_ok = b"\x07Ar\x08)\x80\x10\x102Tv\x98\x07\xf0\xf0\xc0\xc0\x1d\x00\x10\x00\x16\x02\x08\xd0!'\x10\x80\x00\x02\x00\x00\x01\x00\x00\x03\x00\x00\x05\x00\x00\x11\x00\\\x08\x021\x04u\xf0>\x01\x90\x11\x03WX\xb2 \x0b`\x144\xe2\x91\x81\x00\x12>\x80!@\x08\x00\x02\x1f\x00\x04\x02`\x04]\x01\x03\xc1o\x04\xf0\x00\xf0\x00"
    attach_ok, err = parse_NAS_MO(msg_ok)
    print('ok:')
    show(attach_ok)
    show(attach_ok['UENetCap'])

    # problematic
    msg_ko = b'\x17\x07"#\xcb\x05\x07A\x02\x0b\xf6\x02\xf8\x10\x80\x01\x01\x87Q:\x18\x07\xf0\xf0\xc0\xc0\x1d\x00\x10\x00\x16\x02\x07\xd0!\'\x10\x80\x00\x02\x00\x00\x01\x00\x00\x03\x00\x00\x05\x00\x00\x11\x00R\x02\xf8\x10\x00\x01\\\x08\x021\x04u\xf0>\x01\x90\x11\x03WX\xb2 \x0b`\x144\xe2\x91\x81\x00\x12>\x80!@\x08\x00\x02\x1f\x00\x04\x02`\x04]\x01\x03\xe0\xc1o\x04\xf0\x00\xf0\x00'  
    attach_ko, err = parse_NAS_MO(msg_ko)
    print('ko:')
    show(attach_ko)
    show(attach_ko['UENetCap'])
```

here is the error output:
```
Traceback (most recent call last):
  File "C:\Users\alexa\AppData\Local\Programs\Python\Python39\lib\site-packages\pycrate_core\elt.py", line 2133, in __getitem__
    return self._content[self._by_name.index(key)]
ValueError: 'UENetCap' is not in list

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\Users\alexa\Documents\capacabana\capacabana.py", line 5111, in <module>
    capacabana(sys.argv[1:])
  File "c:\Users\alexa\Documents\capacabana\capacabana.py", line 5033, in capacabana
    imei1 = parseLog(infile+'\gnb0.log')
  File "c:\Users\alexa\Documents\capacabana\capacabana.py", line 880, in parseLog
    loadAttach(msg)
  File "c:\Users\alexa\Documents\capacabana\capacabana.py", line 442, in loadAttach
    show(attach_ko['UENetCap'])
  File "C:\Users\alexa\AppData\Local\Programs\Python\Python39\lib\site-packages\pycrate_core\elt.py", line 2135, in __getitem__
    raise(EltErr('{0} [__getitem__] str item: {1}'.format(self._name, err)))
pycrate_core.elt.EltErr: EMMSecProtNASMessage [__getitem__] str item: 'UENetCap' is not in list
sys:1: ResourceWarning: unclosed file <_io.TextIOWrapper name='taibai\\gnb0.log' mode='r' encoding='cp1252'>
```

p1-bmu, 2022-05-05:
In the 2nd message, the `EMMAttachRequest` is wrapped into an `EMMSecProtNASMessage` (the NAS EMM security header). To select the IEs inside the AttachRequest, you need to select it with the proper path:
```
In [1]: m, e = parse_NAS_MO(b'\x17\x07"#\xcb\x05\x07A\x02\x0b\xf6\x02\xf8\x10\x80\x01\x01\x87Q:\x18\x07\xf0\xf0\xc0\xc0\x1d\x00\x10\x00\x16\x02\x07\xd0!\'\x10\x80\x00\x02\x00\x00\x01\x00\x00\x03\x00\x00\x
    ...: 05\x00\x00\x11\x00R\x02\xf8\x10\x00\x01\\\x08\x021\x04u\xf0>\x01\x90\x11\x03WX\xb2 \x0b`\x144\xe2\x91\x81\x00\x12>\x80!@\x08\x00\x02\x1f\x00\x04\x02`\x04]\x01\x03\xe0\xc1o\x04\xf0\x00\xf0\x00')   

In [2]: m['EMMAttachRequest']['UENetCap']                                                                                                                                                                   
Out[2]: <UENetCap : <L : 7><UENetCap : <EEA0 : 1><EEA1_128 : 1><EEA2_128 : 1><EEA3_128 : 1><EEA4 : 0><EEA5 : 0><EEA6 : 0><EEA7 : 0><EIA0 : 1><EIA1_128 : 1><EIA2_128 : 1><EIA3_128 : 1><EIA4 : 0><EIA5 : 0><EIA6 : 0><EIA7 : 0><UEA0 : 1><UEA1 : 1><UEA2 : 0><UEA3 : 0><UEA4 : 0><UEA5 : 0><UEA6 : 0><UEA7 : 0><UCS2 : 1><UIA1 : 1><UIA2 : 0><UIA3 : 0><UIA4 : 0><UIA5 : 0><UIA6 : 0><UIA7 : 0><ProSe_dd : 0><ProSe : 0><H245_ASH : 0><ACC_CSFB : 1><LPP : 1><LCS : 1><X1_SRVCC : 0><NF : 1><ePCO : 0><HC_CP_CIoT : 0><ERw_oPDN : 0><S1U_data : 0><UP_CIoT : 0><CP_CIoT : 0><ProSe_relay : 0><ProSe_dc : 0><spare : 0><spare : 0><spare : 0><spare : 1><spare : 0><spare : 0><spare : 0><MultiDRB : 0><spare [transparent] : 0x>>>
```

abouillot, 2022-05-05:
Thanks, well spotted. I'll addapt the decoding accordingly.

closed 2022-05-05 by abouillot