P1sec/pycrate #193: Stack overflow upon NR RRC Reconfiguration Complete decoding

dbressan2, 2022-05-12:
Hi 

I get a stack overflow with the following code: 

message5 = "08 58 02 00".replace(" ", "")
ul_dcch = RRCNR.NR_RRC_Definitions.UL_DCCH_Message 
ul_dcch.from_uper(bytes.fromhex(message5))

Windows fatal exception: stack overflow
Main thread:
Current thread 0x0000c070 (most recent call first):
  File "C:\Python310\lib\site-packages\pycrate_asn1rt\dictobj.py", line 83 in __iter__
  File "C:\Python310\lib\site-packages\pycrate_asn1rt\asnobj_construct.py", line 790 in _to_asn1
  File "C:\Python310\lib\site-packages\pycrate_asn1rt\asnobj_construct.py", line 795 in _to_asn1
  File "C:\Python310\lib\site-packages\pycrate_asn1rt\asnobj_construct.py", line 125 in _to_asn1
  File "C:\Python310\lib\site-packages\pycrate_asn1rt\asnobj_construct.py", line 795 in _to_asn1
  File "C:\Python310\lib\site-packages\pycrate_asn1rt\asnobj_str.py", line 1342 in _to_asn1

Python version: 3.10.4, Pycrate 0.5.4 

This is supposed to decode as: 

UL-DCCH-Message ::= {
    message: c1: rrcReconfigurationComplete: RRCReconfigurationComplete ::= {
        rrc-TransactionIdentifier: 0
        criticalExtensions: rrcReconfigurationComplete: RRCReconfigurationComplete-IEs ::= {
            nonCriticalExtension: RRCReconfigurationComplete-v1530-IEs ::= {
                nonCriticalExtension: RRCReconfigurationComplete-v1560-IEs ::= {
                    scg-Response: nr-SCG-Response: 
                        NR-RRCReconfigurationComplete ::= {
                            rRCReconfigurationComplete: RRCReconfigurationComplete ::= {
                                rrc-TransactionIdentifier: 0
                                criticalExtensions: rrcReconfigurationComplete: RRCReconfigurationComplete-IEs ::= {
                                }
                            }
                        }

                }
            }
        }
    }
}

Thanks !

dbressan2, 2022-05-12:
> WARNING: This email originated from outside of Qualcomm. Please be wary of any links or attachments, and do not enable macros.
>
> This is not supposed to happen... Do you have a special build of CPython for Windows ?
>
> On the other side, I get a RecursionError when printing the decoded value on my side, meaning there is an issue somewhere in the ASN.1 runtime that is triggered by this RRC NR construct:
>
> In [17]: ul_dcch.from_uper(bytes.fromhex('08580200'))
>
> In [18]: print(ul_dcch.to_asn1())
>
> ---------------------------------------------------------------------------
>
> RecursionError                            Traceback (most recent call last)
>
> <ipython-input-18-96895abf0e02> in <module>
>
> ----> 1 print(ul_dcch.to_asn1())
>
> ~/python/pycrate_asn1rt/asnobj.py in to_asn1(self, val)
>
>    1265             self.set_val(val)
>
>    1266         if self._val is not None:
>
> -> 1267             return self._to_asn1()
>
>    1268         else:
>
>    1269             return None
>
> If I build the following value, I can safely retrieve the expected buffer:
>
> In [68]: v = {'message': ('c1', ('rrcReconfigurationComplete', {'rrc-TransactionIdentifier': 0, 'criticalExtensions': ('rrcReconfigurationComplete', {'nonCriticalExtension': {'nonCriticalExtension': {'scg- Response': ('nr-SCG-Response', b'\x00')}}})}))}
>
> In [69]: ul_dcch.set_val(v)
>
> In [70]: ul_dcch.to_uper()
>
> Out[70]: b'\x08X\x02\x00'
>
> It seems that the RRCReconfigurationComplete object from upper level is getting mixed with the nested one:
>
> In [83]: ul_dcch.from_uper(bytes.fromhex('08580200'))
>
> In [84]: ul_dcch._val
>
> Out[84]:

Thanks for quick reply

This is indeed a recursion error. The stack overflow is encountered with Spyder IDE, maybe it does not properly handle the recursion error. But with another IDE, I get “RecursionError: maximum recursion depth exceeded while calling a Python object”

Thank you very much for your help

Best regards

dB

     {'nonCriticalExtension': {'nonCriticalExtension': {'scg-Response': ('nr-SCG-Response',

In [85]: ul_dcch._val['message'][1][1]['criticalExtensions'][1]['nonCriticalExtension']['nonCriticalExtension']['scg-Response'][1]

Out[85]:

   {'nonCriticalExtension': {'nonCriticalExtension': {'scg-Response': ('nr-SCG-Response',

       (...))}}})})

I need to further investigate why exactly this recursion happens between those 2 RRCReconfigurationComplete objects during the PER decoding.
On your side, it seems your CPython windows build is not that safe, eventually ensure what version you have installed.

p1-bmu, 2022-05-12:
This is actually a true recursion within the 3GPP ASN.1 NR RRC spec, within RRCReconfigurationComplete:
https://github.com/P1sec/pycrate/blob/4c00269b853af95c2f04e2dd31717d16fa45fa8b/pycrate_asn1dir/3GPP_NR_RRC_38331/NR-RRC-Definitions.asn#L788
This is certainly a dangerous construct !

dbressan2, 2022-05-12:
Agreed

However this is commonly used for NR-DC where a first Reconfiguration Complete is meant for primary node (MCG) and the second one (embedded) for the secondary node (SCG).
Would it be possible to handle this case gracefully ? Likewise, an NR RRC Reconfiguration can embed a second RRC Reconfiguration (again in the context of NR-DC).

Thank you very much for your help

Best regards

dB

https://github.com/P1sec/pycrate/blob/4c00269b853af95c2f04e2dd31717d16fa45fa8b/pycrate_asn1dir/3GPP_NR_RRC_38331/NR-RRC-Definitions.asn#L788

p1-bmu, 2022-05-12:
Unfortunately, the way each object's value is assigned within decoding methods does not enable to handle recursion gracefully. I guess I will need to do a large update of most of the decoding methods to manage correctly (and this will takes some time).

dbressan2, 2022-05-12:
Understood.

In the short term I guess I can modify the ASN.1 definitions with some duplication. Something like

RRCReconfigurationComplete-v1560-IEs ::= SEQUENCE {
   * scg-Response CHOICE {
          nr-SCG-Response    OCTET STRING (CONTAINING RRCReconfigurationCompleteSCG),
          eutra-SCG-Response OCTET STRING
     } OPTIONAL,
   * nonCriticalExtension RRCReconfigurationComplete-v1610-IEs OPTIONAL
}

RRCReconfigurationCompleteSCG ::= SEQUENCE {

     rrc-TransactionIdentifier RRC-TransactionIdentifier,

     criticalExtensions CHOICE {

          rrcReconfigurationComplete RRCReconfigurationComplete-IEs,

          criticalExtensionsFuture   SEQUENCE {}

     }

}

Thank you very much for your help

Best regards

dB

p1-bmu, 2022-05-13:
OK, it was finally easier than expected. I just merged a quick branch to handle the decoding of ASN.1 constructed objects, in a way that fits with object's recursion. So, if you git pull the master, your test should not break anymore (neither any recursive constructs).

p1-bmu, 2022-05-13:
Bonus prize, here is a nice NR UL_DCCH_Message buffer that may exercise a bit the gNodeB signaling stack !
```
08591ee41647b70591ed41647b30591ec41647af0591eb41647ab0591ea41647a70591e941647a30591e8416479f0591e7416479b0591e641647970591e541647930591e4416478f0591e3416478b0591e241647870591e141647830591e0416477f0591df416477b0591de41647770591dd41647730591dc416476f0591db416476b0591da41647670591d941647630591d8416475f0591d7416475b0591d641647570591d541647530591d4416474f0591d3416474b0591d241647470591d141647430591d0416473f0591cf416473b0591ce41647370591cd41647330591cc416472f0591cb416472b0591ca41647270591c941647230591c8416471f0591c7416471b0591c641647170591c541647130591c4416470f0591c3416470b0591c241647070591c141647030591c041646ff0591bf41646fb0591be41646f70591bd41646f30591bc41646ef0591bb41646eb0591ba41646e70591b941646e30591b841646df0591b741646db0591b641646d70591b541646d30591b441646cf0591b341646cb0591b241646c70591b141646c30591b041646bf0591af41646bb0591ae41646b70591ad41646b30591ac41646af0591ab41646ab0591aa41646a70591a941646a30591a8416469f0591a7416469b0591a641646970591a541646930591a4416468f0591a3416468b0591a241646870591a141646830591a0416467f05919f416467b05919e416467705919d416467305919c416466f05919b416466b05919a41646670591994164663059198416465f059197416465b05919641646570591954164653059194416464f059193416464b05919241646470591914164643059190416463f05918f416463b05918e416463705918d416463305918c416462f05918b416462b05918a41646270591894164623059188416461f059187416461b05918641646170591854164613059184416460f059183416460b0591824164607059181416460305918041645ff05917f41645fb05917e41645f705917d41645f305917c41645ef05917b41645eb05917a41645e705917941645e305917841645df05917741645db05917641645d705917541645d305917441645cf05917341645cb05917241645c705917141645c305917041645bf05916f41645bb05916e41645b705916d41645b305916c41645af05916b41645ab05916a41645a705916941645a3059168416459f059167416459b05916641645970591654164593059164416458f059163416458b05916241645870591614164583059160416457f05915f416457b05915e416457705915d416457305915c416456f05915b416456b05915a41645670591594164563059158416455f059157416455b05915641645570591554164553059154416454f059153416454b05915241645470591514164543059150416453f05914f416453b05914e416453705914d416453305914c416452f05914b416452b05914a41645270591494164523059148416451f059147416451b05914641645170591454164513059144416450f059143416450b0591424164507059141416450305914041644ff05913f41644fb05913e41644f705913d41644f305913c41644ef05913b41644eb05913a41644e705913941644e305913841644df05913741644db05913641644d705913541644d305913441644cf05913341644cb05913241644c705913141644c305913041644bf05912f41644bb05912e41644b705912d41644b305912c41644af05912b41644ab05912a41644a705912941644a3059128416449f059127416449b05912641644970591254164493059124416448f059123416448b05912241644870591214164483059120416447f05911f416447b05911e416447705911d416447305911c416446f05911b416446b05911a41644670591194164463059118416445f059117416445b05911641644570591154164453059114416444f059113416444b05911241644470591114164443059110416443f05910f416443b05910e416443705910d416443305910c416442f05910b416442b05910a41644270591094164423059108416441f059107416441b05910641644170591054164413059104416440f059103416440b0591024164407059101416440305910041643ff0590ff41643fb0590fe41643f70590fd41643f30590fc41643ef0590fb41643eb0590fa41643e70590f941643e30590f841643df0590f741643db0590f641643d70590f541643d30590f441643cf0590f341643cb0590f241643c70590f141643c30590f041643bf0590ef41643bb0590ee41643b70590ed41643b30590ec41643af0590eb41643ab0590ea41643a70590e941643a30590e8416439f0590e7416439b0590e641643970590e541643930590e4416438f0590e3416438b0590e241643870590e141643830590e0416437f0590df416437b0590de41643770590dd41643730590dc416436f0590db416436b0590da41643670590d941643630590d8416435f0590d7416435b0590d641643570590d541643530590d4416434f0590d3416434b0590d241643470590d141643430590d0416433f0590cf416433b0590ce41643370590cd41643330590cc416432f0590cb416432b0590ca41643270590c941643230590c8416431f0590c7416431b0590c641643170590c541643130590c4416430f0590c3416430b0590c241643070590c141643030590c041642ff0590bf41642fb0590be41642f70590bd41642f30590bc41642ef0590bb41642eb0590ba41642e70590b941642e30590b841642df0590b741642db0590b641642d70590b541642d30590b441642cf0590b341642cb0590b241642c70590b141642c30590b041642bf0590af41642bb0590ae41642b70590ad41642b30590ac41642af0590ab41642ab0590aa41642a70590a941642a30590a8416429f0590a7416429b0590a641642970590a541642930590a4416428f0590a3416428b0590a241642870590a141642830590a0416427f05909f416427b05909e416427705909d416427305909c416426f05909b416426b05909a41642670590994164263059098416425f059097416425b05909641642570590954164253059094416424f059093416424b05909241642470590914164243059090416423f05908f416423b05908e416423705908d416423305908c416422f05908b416422b05908a41642270590894164223059088416421f059087416421b05908641642170590854164213059084416420f059083416420b0590824164207059081416420305908041641ff05907f41641fb05907e41641f705907d41641f305907c41641ef05907b41641eb05907a41641e705907941641e305907841641df05907741641db05907641641d705907541641d305907441641cf05907341641cb05907241641c705907141641c305907041641bf05906f41641bb05906e41641b705906d41641b305906c41641af05906b41641ab05906a41641a705906941641a3059068416419f059067416419b05906641641970590654164193059064416418f059063416418b05906241641870590614164183059060416417f05905f416417b05905e416417705905d416417305905c416416f05905b416416b05905a41641670590594164163059058416415f059057416415b05905641641570590554164153059054416414f059053416414b05905241641470590514164143059050 [... cut ...] 00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```
(you may need to increase the Python's default recursion limit to decode it properly)

dbressan2, 2022-05-16:
Thank you very much

Please note a single level of recursion is needed in practice. This is because 3GPP only supports 2 cell groups, namely Main Cell Group (MCG) and Secondary Cell Group (SCG). This is unlikely to change.

Best regards

dB

closed 2022-06-09 by p1-bmu