P1sec/pycrate #195: Iterating through a SEQ OF with get_at method

dbressan2, 2022-05-17:
Hi 

I am attempting to get an octet string object corresponding to each instance of ueCapabilityRAT-Container

I am getting unexpected results with get_at where it always returns the octet string corresponding to last instance of ueCapabilityRAT-Container, whereas I would like to iterate through each instance. 
How can we use get_at for SEQ OF ? 

I am trying the following code: 

message0 = """
3e 03 08 19 2d db 80 50 30 9f bf 7e 0c 1f 83 f0
7e 0f df 83 76 26 80 08 0d cf 64 00 00 00 01 78
68 0c 0a 0c 77 c0 00 00 00 b2 f0 03 00 24 10 86
00 60 84 30 03 00 30 40 18 20 0c 20 84 30 42 18
21 0c 10 86 08 43 00 30 80 10 c2 00 43 04 01 82
00 c0 0c 00 c0 09 68 3e aa 30 e0 b8 18 40 00 82
07 f8 00 20 80 fe 10 00 ff 00 04 00 70 80 01 05
08 20 81 84 00 09 20 ff 80 02 0b f8 41 03 fc 00
12 40 ff 00 04 50 7f c0 01 01 fc 20 82 fe 10 3f
c0 01 04 2f f0 00 41 7f 08 2f e1 04 1f f0 00 49
7f 08 20 bf 84 9f f0 00 41 07 fc 00 11 5f c2 08
2f e1 17 fc 00 10 40 ff 00 04 d0 bf c0 01 01 fc
20 bf 84 10 9f c2 07 f8 00 20 bf 84 10 7f c0 01
01 fc 24 82 fe 10 3f c0 01 24 0c 20 10 49 03 fc
04 12 40 ff 00 84 52 e0 fc 1f 83 f0 7e 0f c1 f8
3f 07 e0 fc 1f 83 f0 7e 0f c1 f8 3f 07 e0 fc 1f
83 f0 7e 0f c1 f8 3f 07 ed 9d 00 00 11 97 08 b8
00 00 e5 c0 00 40 40 40 80 41 81 00 00 50 14 04
04 04 04 04 14 19 01 05 06 40 42 40 50 14 04 14
05 01 05 01 40 41 40 50 10 50 14 04 04 04 24 19
01 40 42 41 90 14 04 14 19 01 05 06 40 40 00 40
40 40 66 80 03 e6 57 9a 04 01 48 7f f9 a5 2e f1
c0 3a 00 16 08 00 64 92 00 00 40 00 a2 09 73 40
00 a1 e1 00 00 00 21 00 00 01 02 08 20 80 00 20
82 18 20 82 08 20 80 18 00 00 00 51 90 00 00 00
40 02 20 10 50 42 05 81 0c e1 20 10 00 07 4f 5a
13 02 00 02 01 40 48 24 c5 80 02 60 d4 0c 13 60
18 73 e6 18 f2 85 81 c1 30 08 0c 41 00 3a 0d 85
18 1a c1 18 43 41 02 0f 94 02 2c b6 4c 55 0d 0e
7c 47 1e 81 9f 60 4f 8c 73 e6 18 f2 85 81 c0 e1
e0 b8 00 27 80 0e 00 09 e0 00 38 01 c0 00 20 0a
01 47 27 80 22 03 01 f0 28 01 00 00 00 00 80 00
01 00 40 00 01 00 20 00 00 c0 10 00 00 60 08 00
00 40 04 00 00 28 02 00 00 18 01 00 00 04 00 80
00 07 00 40 00 04 01 0c a9 bc a9 ac 0c ac 0d ac
99 ac a5 ac 8d ac 04 ac 08 a0 48 20 00 01 2c a4
a0 80 00 0c b2 92 82 00 00 52 ca 48 08 00 01 cb
29 20 20 00 09 2c a4 80 80 00 2c b2 92 02 00 00
d2 ca 48 08 00 03 cb 29 20 20 00 11 2c a4 80 80
00 4c b2 92 01 2a 55 57 4a aa a8 1c 55 2a 8b a5
51 54 9a 2a 94 c5 52 38 aa 03 15 40 a2 80 87 40
2a 81 40 e0 70 28 14 0a 05 02 81 40 a0 08 50 18
12 0a 12 32 00 62 5a 10 7c 04 a1 60 40 1c 28 00
00 00 44 00 40 14 02 80 b0 03 a0 30 a7 42 d8 e7
c3 10 00 08 00 08 02 02 00 80 28 00 
""".replace(" ", "").replace("\n", "")
dcch = lte_g80.EUTRA_RRC_Definitions.UL_DCCH_Message
dcch.from_uper(bytes.fromhex(message0))
print (dcch.to_asn1())
container_list = ['message',  'c1', 'ueCapabilityInformation',  'criticalExtensions',  'c1', 'ueCapabilityInformation-r8',
                                              'ue-CapabilityRAT-ContainerList']

print (dcch.to_asn1())
num_tech = 3
for tech in range(num_tech):
                                
    tech_rat_container = dcch.get_val_at(container_list + [tech])
    rat_info = tech_rat_container['rat-Type']
    print(rat_info)
    # this works 
    container = tech_rat_container['ueCapabilityRAT-Container']
    print (container)
    # this does not work and returns last ueCapabilityRAT-Container regardless of tech value 
    oct_str = dcch.get_at(container_list + [tech, 'ueCapabilityRAT-Container']) 
    print ("oct str for rat info {0} = {1}".format(rat_info, oct_str._val.hex()))

Thanks

p1-bmu, 2022-05-18:
`get_at()` is for selecting a sub-object from a constructed one. Here, you better want to use `get_val_at()` to select a sub-value from a value assigned to the constructed object.

dbressan2, 2022-05-18:
> WARNING: This email originated from outside of Qualcomm. Please be wary of any links or attachments, and do not enable macros.

Thanks

My problem is that I am trying to link object to invoke NR decoding for ueCapabilityRAT-Container for NR rat

Something like:

     if rat_info == "nr":
         print(oct_str._val.hex())
         _cont_UE_NR_Capability = SEQ(name='UE-NR-Capability', mode=MODE_TYPE,
                                      typeref=ASN1RefType(('NR-RRC-Definitions', 'UE-NR-Capability')))
         oct_str._const_cont =  _cont_UE_NR_Capability
         oct_str._const_cont_enc = None
         extensions = [_cont_UE_NR_Capability]
         lte_g80.EUTRA_RRC_Definitions._all_.extend(extensions)
         init_modules(lte_g80.EUTRA_RRC_Definitions)
         dcch.from_uper(bytes.fromhex(message0))

So I somehow need to select the sub-object corresponding to ueCapabilityRAT-Container for rat-Type = nr. Any way I can achieve this ?

Thank you very much for your help

Best regards

dB

dbressan2, 2022-05-19:
Hi

For clarity, here is the code I am trying to write:

message0 = """
3e 03 08 19 2d db 80 50 30 9f bf 7e 0c 1f 83 f0
7e 0f df 83 76 26 80 08 0d cf 64 00 00 00 01 78
68 0c 0a 0c 77 c0 00 00 00 b2 f0 03 00 24 10 86
00 60 84 30 03 00 30 40 18 20 0c 20 84 30 42 18
21 0c 10 86 08 43 00 30 80 10 c2 00 43 04 01 82
00 c0 0c 00 c0 09 68 3e aa 30 e0 b8 18 40 00 82
07 f8 00 20 80 fe 10 00 ff 00 04 00 70 80 01 05
08 20 81 84 00 09 20 ff 80 02 0b f8 41 03 fc 00
12 40 ff 00 04 50 7f c0 01 01 fc 20 82 fe 10 3f
c0 01 04 2f f0 00 41 7f 08 2f e1 04 1f f0 00 49
7f 08 20 bf 84 9f f0 00 41 07 fc 00 11 5f c2 08
2f e1 17 fc 00 10 40 ff 00 04 d0 bf c0 01 01 fc
20 bf 84 10 9f c2 07 f8 00 20 bf 84 10 7f c0 01
01 fc 24 82 fe 10 3f c0 01 24 0c 20 10 49 03 fc
04 12 40 ff 00 84 52 e0 fc 1f 83 f0 7e 0f c1 f8
3f 07 e0 fc 1f 83 f0 7e 0f c1 f8 3f 07 e0 fc 1f
83 f0 7e 0f c1 f8 3f 07 ed 9d 00 00 11 97 08 b8
00 00 e5 c0 00 40 40 40 80 41 81 00 00 50 14 04
04 04 04 04 14 19 01 05 06 40 42 40 50 14 04 14
05 01 05 01 40 41 40 50 10 50 14 04 04 04 24 19
01 40 42 41 90 14 04 14 19 01 05 06 40 40 00 40
40 40 66 80 03 e6 57 9a 04 01 48 7f f9 a5 2e f1
c0 3a 00 16 08 00 64 92 00 00 40 00 a2 09 73 40
00 a1 e1 00 00 00 21 00 00 01 02 08 20 80 00 20
82 18 20 82 08 20 80 18 00 00 00 51 90 00 00 00
40 02 20 10 50 42 05 81 0c e1 20 10 00 07 4f 5a
13 02 00 02 01 40 48 24 c5 80 02 60 d4 0c 13 60
18 73 e6 18 f2 85 81 c1 30 08 0c 41 00 3a 0d 85
18 1a c1 18 43 41 02 0f 94 02 2c b6 4c 55 0d 0e
7c 47 1e 81 9f 60 4f 8c 73 e6 18 f2 85 81 c0 e1
e0 b8 00 27 80 0e 00 09 e0 00 38 01 c0 00 20 0a
01 47 27 80 22 03 01 f0 28 01 00 00 00 00 80 00
01 00 40 00 01 00 20 00 00 c0 10 00 00 60 08 00
00 40 04 00 00 28 02 00 00 18 01 00 00 04 00 80
00 07 00 40 00 04 01 0c a9 bc a9 ac 0c ac 0d ac
99 ac a5 ac 8d ac 04 ac 08 a0 48 20 00 01 2c a4
a0 80 00 0c b2 92 82 00 00 52 ca 48 08 00 01 cb
29 20 20 00 09 2c a4 80 80 00 2c b2 92 02 00 00
d2 ca 48 08 00 03 cb 29 20 20 00 11 2c a4 80 80
00 4c b2 92 01 2a 55 57 4a aa a8 1c 55 2a 8b a5
51 54 9a 2a 94 c5 52 38 aa 03 15 40 a2 80 87 40
2a 81 40 e0 70 28 14 0a 05 02 81 40 a0 08 50 18
12 0a 12 32 00 62 5a 10 7c 04 a1 60 40 1c 28 00
00 00 44 00 40 14 02 80 b0 03 a0 30 a7 42 d8 e7
c3 10 00 08 00 08 02 02 00 80 28 00
""".replace(" ", "").replace("\n", "")
dcch = lte_g80.EUTRA_RRC_Definitions.UL_DCCH_Message
dcch.from_uper(bytes.fromhex(message0))
print (dcch.to_asn1())
container_list = ['message',  'c1', 'ueCapabilityInformation',  'criticalExtensions',  'c1', 'ueCapabilityInformation-r8',
                                              'ue-CapabilityRAT-ContainerList']

print (dcch.to_asn1())
num_tech = 3
extensions = []
for tech in range(num_tech):

    tech_rat_container = dcch.get_val_at(container_list + [tech])
    rat_info = tech_rat_container['rat-Type']
    print(rat_info)
    container = tech_rat_container['ueCapabilityRAT-Container']
    print (container)
    # this does not work as it always returns the octet string for last 'ueCapabilityRAT-Container', regardless of tech value
    oct_str = dcch.get_at(container_list + [tech, 'ueCapabilityRAT-Container'])
    print ("oct str for rat info {0} = {1}".format(rat_info, oct_str._val.hex()))
    # print (oct_str)
    # print (oct_str.get_val_paths())
    if rat_info == "nr":
        print(oct_str._val.hex())
        _cont_UE_NR_Capability = SEQ(name='UE-NR-Capability', mode=MODE_TYPE,
                                      typeref=ASN1RefType(('NR-RRC-Definitions', 'UE-NR-Capability')))
        oct_str._const_cont =  _cont_UE_NR_Capability
        oct_str._const_cont_enc = None
        # extensions = [_cont_UE_NR_Capability]
        lte_g80.EUTRA_RRC_Definitions._all_.extend(extensions)
        init_modules(lte_g80.EUTRA_RRC_Definitions)
        dcch.from_uper(bytes.fromhex(message0))

Thank you very much for your help

Best regards

dB

p1-bmu, 2022-06-27:
The 1st level sub-object of a SEQUENCE OF (or SET OF) is not named in ASN.1. It's named `_item_` in pycrate (because we still need one programmatically). And you don't need to pass any specific naming when selecting it with `get_at()`.

See e.g. at the end of this section of the wiki: https://github.com/P1sec/pycrate/wiki/Using-the-pycrate-asn1-runtime#rrc-3g

dbressan2, 2022-06-27:
HI

I still do not understand how it is supposed to work. get_val_paths returns some paths and values for an object, but trying to access one path with get_at gives unexpected results.

container_list = ['message',  'c1', 'ueCapabilityInformation',  'criticalExtensions',  'c1', 'ueCapabilityInformation-r8',
                                              'ue-CapabilityRAT-ContainerList']

containers = dcch.get_at(container_list)
containers.get_val_paths()
Out[22]:
[([0, 'rat-Type'], 'eutra'),
([0, 'ueCapabilityRAT-Container'],
  b'\xdd\xb8\x05\x03\t\xfb\xf7\xe0\xc1\xf8?\x07\xe0\xfd\xf87bh\x00\x80\xdc\xf6@\x00\x00\x00\x17\x86\x80\xc0\xa0\xc7|\x00\x00\x00\x0b/\x000\x02A\x08`\x06\x08C\x000\x03\x04\x01\x82\x00\xc2\x08C\x04!\x82\x10\xc1\x08`\x840\x03\x08\x01\x0c \x040@\x18 \x0c\x00\xc0\x0c\x00\x96\x83\xea\xa3\x0e\x0b\x81\x84\x00\x08 \x7f\x80\x02\x08\x0f\xe1\x00\x0f\xf0\x00@\x07\x08\x00\x10P\x82\x08\x18@\x00\x92\x0f\xf8\x00 \xbf\x84\x10?\xc0\x01$\x0f\xf0\x00E\x07\xfc\x00\x10\x1f\xc2\x08/\xe1\x03\xfc\x00\x10B\xff\x00\x04\x17\xf0\x82\xfe\x10A\xff\x00\x04\x97\xf0\x82\x0b\xf8I\xff\x00\x04\x10\x7f\xc0\x01\x15\xfc \x82\xfe\x11\x7f\xc0\x01\x04\x0f\xf0\x00M\x0b\xfc\x00\x10\x1f\xc2\x0b\xf8A\t\xfc \x7f\x80\x02\x0b\xf8A\x07\xfc\x00\x10\x1f\xc2H/\xe1\x03\xfc\x00\x12@\xc2\x01\x04\x90?\xc0A$\x0f\xf0\x08E.\x0f\xc1\xf8?\x07\xe0\xfc\x1f\x83\xf0~\x0f\xc1\xf8?\x07\xe0\xfc\x1f\x83\xf0~\x0f\xc1\xf8?\x07\xe0\xfc\x1f\x83\xf0~\xd9\xd0\x00\x01\x19p\x8b\x80\x00\x0e\\\x00\x04\x04\x04\x08\x04\x18\x10\x00\x05\x01@@@@@***@***.***@P\x10P\x14\x04\x14\x05\x01\x05\x01@@@***@***.***\x90\x10Pd\x04\x00\x04\x04\x04\x06h\x00>ey\xa0@\x14\x87\xff\x9aR\xef\x1c\x03\xa0\x01`\x80\x06I \x00\x04\x00\n \x974\x00\n\x1e\x10\x00\x00\x02\x10\x00\x00\x10 \x82\x08\x00\x02\x08!\x82\x08 \x82\x08\x01\x80\x00\x00\x05\x19\x00\x00\x00\x04\x00"\x01\x05\x04 '),
([1, 'rat-Type'], 'nr'),
([1, 'ueCapabilityRAT-Container'],
  b'\xe1 ***@***.***$\xc5\x80\x02`\xd4\x0c\x13`\x18s\xe6\x18\xf2\x85\x81\xc10\x08\x0cA\x00:\r\x85\x18\x1a\xc1\x18CA\x02\x0f\x94\x02,\xb6LU\r\x0e|G\x1e\x81\x9f`O\x8cs\xe6\x18\xf2\x85\x81\xc0\xe1\xe0\xb8\x00\'\x80\x0e\x00\t\xe0\x008\x01\xc0\x00 \n\x01G\'\x80"\x03\x01\xf0(\x01\x00\x00\x00\x00\x80\x00\x01\x00@\x00\x01\x00 \x00\x00\xc0\x10\x00\x00`\x08\x00\x00@\x04\x00\x00(\x02\x00\x00\x18\x01\x00\x00\x04\x00\x80\x00\x07\x00@\x00\x04\x01\x0c\xa9\xbc\xa9\xac\x0c\xac\r\xac\x99\xac\xa5\xac\x8d\xac\x04\xac\x08\xa0H \x00\x01,\xa4\xa0\x80\x00\x0c\xb2\x92\x82\x00\x00R\xcaH\x08\x00\x01\xcb)  \x00\t,\xa4\x80\x80\x00,\xb2\x92\x02\x00\x00\xd2\xcaH\x08\x00\x03\xcb)  \x00\x11,\xa4\x80\x80\x00L\xb2\x92\x01*UWJ\xaa\xa8\x1cU*\x8b\xa5QT\x9a*\x94\xc5R8\xaa\x03\x15@\xa2\x80\x87@*\x81@\xe0p(\x14\n\x05\x02\x81@\xa0\x08P\x18\x12\n\x122\x00'),
([2, 'rat-Type'], 'eutra-nr'),
([2, 'ueCapabilityRAT-Container'],
  b'\xa1\x07\xc0J\x16\x04\x01\xc2\x80\x00\x00\x04@\x04\x01@(\x0b\x00:\x03\nt-\x8e|1\x00\x00\x80\x00\x80  \x08\x02\x80')]

containers.get_at([1, 'rat-Type']).get_val()

Out[24]: 'eutra-nr' # why ? Expected value = ‘nr’

containers.get_at([5, 'rat-Type']).get_val()
Out[25]: 'eutra-nr' # why ? this is an invalid paths.

Thank you very much for your help

Best regards

dB

p1-bmu, 2022-06-27:
Is this the same issue as in https://github.com/P1sec/pycrate/issues/203 ? If yes, I suggest you close one of the too. Looking at what you do:
```
containers.get_at([1, 'rat-Type']).get_val()
```
I guess this is the same. And again, `get_at()` if for object, and there is not such _itemized_ / different sub-object within a SEQUENCE OF, only a single object, but eventually several different values to be retrieved with `get_val_at()`.

Eventually, go through the source code (https://github.com/P1sec/pycrate/blob/5fa6dda9d0ff8754225e78fc92dbbd9da3dc9546/pycrate_asn1rt/asnobj.py#L960). And if you are still lost, please ask for support internally at Qualcomm first.

dbressan2, 2022-06-28:
Hi 
Thanks for comments. The issue is indeed similar as #203, I am closing this one and we can track the discussion further with #203

closed 2022-06-28 by dbressan2