P1sec/pycrate #196: WITH COMPONENTS

turabek-cc, 2022-05-18:
I am trying to compile IEEE 1609.2 specification. The line with `psid (CrlPsid )` is failing with the following
```
$ :~/quickasn1test/test_pycrate$ pycrate_asn1compile.py -i 1609dot2.asn -o 1609dot2 -j
[proc] [1609dot2.asn] module IEEE1609dot2BaseTypes (oid: [1, 3, 111, 2, 1609, 2, 1, 2, 2]): 72 ASN.1 assignments found
[proc] [1609dot2.asn] module IEEE1609dot2 (oid: [1, 3, 111, 2, 1609, 2, 1, 1, 2]): 36 ASN.1 assignments found
[proc] [1609dot2.asn] module IEEE1609dot2-Peer2Peer (oid: [1, 3, 111, 2, 1609, 2, 2, 1, 2]): 3 ASN.1 assignments found
[proc] [1609dot2.asn] module IEEE1609dot2CrlBaseTypes (oid: [1, 3, 111, 2, 1609, 2, 3, 2, 2]): 17 ASN.1 assignments found
[proc] [1609dot2.asn] module IEEE1609dot2Crl (oid: [1, 3, 111, 2, 1609, 2, 3, 1, 2]): 3 ASN.1 assignments found
[proc] [1609dot2.asn] module IEEE1609dot2CrlSsp (oid: [1, 3, 111, 2, 1609, 2, 3, 3, 2]): 4 ASN.1 assignments found
[proc] [1609dot2.asn] module EtsiTs102941Crl (oid: [1, 3, 111, 2, 1609, 2, 3, 100, 1, 2]): 3 ASN.1 assignments found
--- compilation cycle ---
--- compilation cycle ---
Traceback (most recent call last):
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 6212, in __parse_set_from_const_comp
    comp_exp = comp.expand()
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/setobj.py", line 118, in expand
    raise(ASN1Err('{0!r}: unable to expand infinite range'.format(self)))
pycrate_asn1c.err.ASN1Err: ASN1RangeInt(0..None): unable to expand infinite range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/turabek/quickasn1test/test_pycrate/venv/bin/pycrate_asn1compile.py", line 267, in <module>
    sys.exit(main())
  File "/home/turabek/quickasn1test/test_pycrate/venv/bin/pycrate_asn1compile.py", line 219, in main
    compile_text(txt, **ckw)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnproc.py", line 257, in compile_text
    compile_modules(remain)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnproc.py", line 945, in compile_modules
    ObjNew = asnobj_compile(Obj)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnproc.py", line 879, in asnobj_compile
    text = Obj.parse_def(Obj._text_def)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 1716, in parse_def
    rest = self._parse_const(text)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 3976, in _parse_const
    self._parse_const_withcomps(const)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 4284, in _parse_const_withcomps
    self.__parse_const_withcomps_comp(rc, const_comp)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 4376, in __parse_const_withcomps_comp
    comp = Comp._parse_const(comp)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 3976, in _parse_const
    self._parse_const_withcomps(const)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 4284, in _parse_const_withcomps
    self.__parse_const_withcomps_comp(rc, const_comp)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 4376, in __parse_const_withcomps_comp
    comp = Comp._parse_const(comp)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 3976, in _parse_const
    self._parse_const_withcomps(const)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 4284, in _parse_const_withcomps
    self.__parse_const_withcomps_comp(rc, const_comp)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 4376, in __parse_const_withcomps_comp
    comp = Comp._parse_const(comp)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 3976, in _parse_const
    self._parse_const_withcomps(const)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 4284, in _parse_const_withcomps
    self.__parse_const_withcomps_comp(rc, const_comp)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 4376, in __parse_const_withcomps_comp
    comp = Comp._parse_const(comp)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 3976, in _parse_const
    self._parse_const_withcomps(const)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 4284, in _parse_const_withcomps
    self.__parse_const_withcomps_comp(rc, const_comp)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 4376, in __parse_const_withcomps_comp
    comp = Comp._parse_const(comp)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 3964, in _parse_const
    self._parse_const_val(const)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 4033, in _parse_const_val
    self.parse_set(const['text'])
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 5844, in parse_set
    rest = self._parse_set_comp(rv, val, dom)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 6051, in _parse_set_comp
    objval = ObjProxy.__parse_set_from_const()
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 6189, in __parse_set_from_const
    self.__parse_set_from_const_comp(val, root)
  File "/home/turabek/quickasn1test/test_pycrate/venv/lib/python3.8/site-packages/pycrate_asn1c/asnobj.py", line 6214, in __parse_set_from_const_comp
    raise(ASN1NotSuppErr('{0}: unable to expand range, {1}'\
pycrate_asn1c.err.ASN1NotSuppErr: _set_psid: unable to expand range, ASN1RangeInt(0..None): unable to expand infinite range
```

Here is the excerpt from specification. 
```
Psid ::= INTEGER (0..MAX)

CrlPsid ::= Psid(256)

SecuredCrl ::= Ieee1609Dot2Data (WITH COMPONENTS {..., 
  content (WITH COMPONENTS {
    signedData  (WITH COMPONENTS {..., 
      tbsData (WITH COMPONENTS {
        payload (WITH COMPONENTS {..., 
          data (WITH COMPONENTS {...,
             content (WITH COMPONENTS {
                unsecuredData (CONTAINING CrlContents)
            })
          })
        })
        ,
        headerInfo (WITH COMPONENTS {..., 
          psid (CrlPsid ),
          generationTime ABSENT,
          expiryTime ABSENT,
          generationLocation ABSENT,
          p2pcdLearningRequest ABSENT,
          missingCrlIdentifier ABSENT,
          encryptionKey ABSENT
        })
      })
    })
  })
})
```

If I change `psid (CrlPsid )` to `psid (256)` compilation is successful. However, I am not sure if it will work as expected or not. Is there some workaround for this situation?

p1-bmu, 2022-05-18:
I guess this:
```
Psid ::= INTEGER (0..MAX)
CrlPsid ::= Psid(256)
psid ::= Psid (CrlPsid)
```
is not properly supported by the pycrate's compiler. These ITS ASN.1 specifications are so convoluted...

p1-bmu, 2022-06-10:
@turabek-cc : which spec are you using exactly ? As there is an ETSI_ITS_IEEE1609_2 in pycrate, which compiles fine: https://github.com/P1sec/pycrate/tree/master/pycrate_asn1dir/ETSI_ITS_IEEE1609_2/.

turabek-cc, 2022-06-13:
@p1-bmu I am using the files from https://standards.ieee.org/wp-content/uploads/import/download/1609.2.1-2020_downloads.zip .  Specifically, the following files:

- 1609dot2-P2P.asn  
- 1609dot2-base-types.asn  
- 1609dot2-schema.asn  
- EtsiTs102941Crl.asn  
- crl-base-types.asn  
- crl-protocol.asn  
- crl-ssp.asn

p1-bmu, 2022-06-15:
OK, I think the last commit should have solved the issue. Please try to recompile your modules after a fresh `git pull` and let me know.

turabek-cc, 2022-06-16:
I confirm that the last commit solved the issue. I am able to compile the files listed above without any modifications.

`[proc] [1609dot2.asn] module IEEE1609dot2BaseTypes (oid: [1, 3, 111, 2, 1609, 2, 1, 2, 2]): 72 ASN.1 assignments found
[proc] [1609dot2.asn] module IEEE1609dot2 (oid: [1, 3, 111, 2, 1609, 2, 1, 1, 2]): 36 ASN.1 assignments found
[proc] [1609dot2.asn] module IEEE1609dot2-Peer2Peer (oid: [1, 3, 111, 2, 1609, 2, 2, 1, 2]): 3 ASN.1 assignments found
[proc] [1609dot2.asn] module IEEE1609dot2CrlBaseTypes (oid: [1, 3, 111, 2, 1609, 2, 3, 2, 2]): 17 ASN.1 assignments found
[proc] [1609dot2.asn] module IEEE1609dot2Crl (oid: [1, 3, 111, 2, 1609, 2, 3, 1, 2]): 3 ASN.1 assignments found
[proc] [1609dot2.asn] module IEEE1609dot2CrlSsp (oid: [1, 3, 111, 2, 1609, 2, 3, 3, 2]): 4 ASN.1 assignments found
[proc] [1609dot2.asn] module EtsiTs102941Crl (oid: [1, 3, 111, 2, 1609, 2, 3, 100, 1, 2]): 3 ASN.1 assignments found
--- compilation cycle ---
--- compilation cycle ---
--- compilation cycle ---
WNG: IEEE1609dot2._set_Certificate, unprocessed set of values from constraint, _set_Certificate
WNG: IEEE1609dot2._set_Certificate, unprocessed set of values from constraint, _set_Certificate
--- verifications ---
[proc] ASN.1 modules processed: ['IEEE1609dot2BaseTypes', 'IEEE1609dot2', 'IEEE1609dot2-Peer2Peer', 'IEEE1609dot2CrlBaseTypes', 'IEEE1609dot2Crl', 'IEEE1609dot2CrlSsp', 'EtsiTs102941Crl']
[proc] ASN.1 objects compiled: 131 types, 0 sets, 0 values
[proc] done`

Thank you!

closed 2022-06-16 by turabek-cc