P1sec/pycrate #197: Decoding NR5G RRC Message DL-DCCH RRCReconfiguration

mamkarim, 2022-06-09:
I am trying to decode "NR5G RRC Message DL-DCCH RRCReconfiguration", but getting empty result. Below is the code -

from pycrate_asn1dir import RRCNR

NR_RRC_Reconfig = RRCNR.NR_RRC_Definitions.DL_DCCH_Message
test1 = 'C883505C40718E721BF18683B23207D00812A0F10C041ADD427318C2D26928009A4D12F0000150B193F9A8282AC028806D00D6021C04280A1840000543AB8C9FCCA9880901A99EFFA580127985370ADE1CDC49B8B371A6E4CDCA9B9737326D354C185A3BF8A42C07601249C99EDBE002038019300CB3DC41844DC500C0000802016AC028806D00D6021C04280A0DAC048FFFFF80000008500041030817A8A8771C0300002008372043000028144722520C04A200C22C004808009206004482801920E008484802921600C4C080E4C18104C28124C38144C48164C58A8A06C3521908A0146891C32E7B8832A6202606A67BFE960049E614DC2B78737126E2CDC69B93372A6E5CDCC9B4D53061686AAA93038002081840A183D38A163068E1E408822089244B009F0110A018101A788500C127806428060C089D21403089D0210A018502A708500C32700A428061C0C9B2140310005096A018904A5D8D06293E0261418B01A798506313C0361418D022758506393A0461418F02A7185064138056141910326D8506492E09634102591813211D159E26F0A908010964769A6B00142000001001821DCC29B856F0E6E24DC59B8D37266E54DCB9B99378FFD6B5B5AD6F281C0000205F040100010000F84003090038010180CA8080CC5555018600228D4008C4030D895C080421001185461A180CE0602208008C2A30D0C067040108400461518686433828088200230A8C343219C1810210011855AEA180EE0E08208008C2AD750C07708040840046156BA8643B848208200230AB5D4321DC280829001185461A1810E1604248008C2A30D0C0870C020A400461518686443868109200230A8C343221C3820290011855AEA1812E1E10248008C2AD750C09710080A40046156BA8644B888409200230AB5D43225C480431001185461A1814E2602288008C2A30D0C0A714010C4004615186864538A808A200230A8C343229C5810310011855AEA1816E2E08288008C2AD750C0B718040C40046156BA8645B8C820A200230AB5D4322C600000050C302030405A88183038404D450C282C3036A3061C1E202351C312131415A9018B0B8C0CC0E0A600465894000000001400420C4144002000C17000C1440301200CC068043022014C0A806303201CC0E8083025A0000B2460112788FFE107F87A4020851226092DEA0'

NR_RRC_Reconfig.from_uper(unhexlify(test1))
print(NR_RRC_Reconfig())
>> {'message': ('messageClassExtension', {})}

I tried the string provided for issue #193 and it works.

NR_RRC_Reconfig = RRCNR.NR_RRC_Definitions.UL_DCCH_Message
message5 = "08 58 02 00".replace(" ", "")
NR_RRC_Reconfig.from_uper(unhexlify(message5))
print(NR_RRC_Reconfig())
>>{'message': ('c1', ('rrcReconfigurationComplete', {'rrc-TransactionIdentifier': 0, 'criticalExtensions': ('rrcReconfigurationComplete', {'nonCriticalExtension': {'nonCriticalExtension': {'scg-Response': ('nr-SCG-Response', ('RRCReconfigurationComplete', {'rrc-TransactionIdentifier': 0, 'criticalExtensions': ('rrcReconfigurationComplete', {...})}))}}})}))}

Appreciate if can get little guidance, tried to look at the wiki, but found nothing for 5G.

p1-bmu, 2022-06-09:
Where does this buffer come from ? Are you sure it corresponds exactly to an NR RRC DL-DCCH-Message ? From what it looks, I would say this may not be an exact or complete DL-DCCL-Message.

mamkarim, 2022-06-09:
I took it from a drive test log. I will try with a simpler message and will try to compare. Thanks a lot for the comment.

mamkarim, 2022-06-13:
I am verifying using logs. Will keep you posted.

closed 2022-06-13 by mamkarim