P1sec/pycrate #198: Type mismatch error

mbrehler, 2022-06-20:
Hi,

When defining a private message IE in X2AP-PDU-Contents.asn:

```
PrivateMessage-IEs X2AP-PRIVATE-IES ::= {
	{ ID id-MeNB-UE-X2AP-ID						CRITICALITY reject	TYPE UE-X2AP-ID											PRESENCE mandatory},
	...
}
```
(code also at https://github.com/P1sec/pycrate/compare/master...mbrehler:type_issue ), we get a compile error we don't understand:

Exception has occurred: ASN1ProcTextErr
id: value identifier id-MeNB-UE-X2AP-ID, type mismatch, INTEGER instead of CHOICE

Any pointers on how to fix would be welcome. 

Thanks,
Matthias

p1-bmu, 2022-06-27:
For `X2AP-PRIVATE-IES`, the `ID` component is of type `PrivateIE-ID`, which is a CHOICE:
```
PrivateIE-ID	::= CHOICE {
	local				INTEGER (0.. maxPrivateIEs),
	global				OBJECT IDENTIFIER
}
```

Here, you pass an INTEGER instead of a CHOICE as value, which the compiler can't accept.

mbrehler, 2022-06-28:
We agree, sorry we (and another, commercial ASN.1 compiler) missed this. Thanks for taking a look and setting us straight.

closed 2022-06-28 by mbrehler