P1sec/pycrate #202: send isup message

AmroAbdulla, 2022-06-21:
how can I work with isup to send iam message to Network mobile

p1-bmu, 2022-06-27:
Sorry, I am not sure to understand the question. If you want to encode an ISUP IAM message, then, it's a matter of filling the `InitialAddress` object with your expected values:
```
In [1]: from pycrate_mobile.ISUP import *                                                                                                                                                                    

In [2]: print(InitialAddress().show())                                                                                                                                                                       
### InitialAddress ###
 <CIC : 0>
 <Type : 1 (Initial address)>
 ### NatureConnectionInd ###
  <spare : 0x0>
  <EchoControlDeviceInd : 0 (incoming echo control device not included)>
  <ContinuityCheckInd : 0 (continuity check not required)>
  <SatelliteInd : 0 (no satellite circuit in the connection)>
 ### ForwardCallInd ###
  <ISDNUserPartPrefInd : 0 (ISDN user part preferred all the way)>
  <ISDNUserPartInd : 0 (ISDN user part not used all the way)>
  <EtoEInfoInd : 0 (no end-to-end information available)>
  <InterworkingInd : 0 (no interworking encountered (Signalling System No. 7 all the way))>
  <EtoEMethodInd : 0 (no end-to-end method available (only link-by-link method available))>
  <InternatCallInd : 0 (call to be treated as a national call)>
  <ReservedNationalUse : 0>
  <spare : 0>
  <SCCPMethodInd : 0 (no indication)>
  <ISDNAccessInd : 0 (originating access non-ISDN)>
 <CallingPartyCat : 0 (calling party's category unknown at this time (national use))>
 <TransmissionMediumReq : 0 (speech)>
 ### Pointers ###
  <Ptr0 : 2>
  <Ptr1 : 0>
 ### CalledPartyNum ###
  <Len : 2>
  ### CalledPartyNum ###
   <OE : 0 (even number of address signals)>
   <NAI : 1 (subscriber number (national use))>
   <INN : 0 (routing to internal network number allowed)>
   <NumPlan : 0 (spare)>
   <spare : 0x0>
   <Num : >
 ### Opt ###
```
Then, calling the `to_bytes()` method to encode it into a buffer.

If your question is related to something else, e.g., where you can connect to a telecom routing point, then I can't answer.

AmroAbdulla, 2022-06-30:
i have my own telecom routing point but i want to build a sigtran stack(sctp , M3ua , isup) and send it to my telecom routing point

p1-bmu, 2022-07-01:
So you can use Linux + https://github.com/P1sec/pysctp for SCTP
https://github.com/P1sec/pycrate/blob/master/pycrate_mobile/SIGTRAN.py for M2PA and MTP3
https://github.com/P1sec/pycrate/blob/master/pycrate_mobile/ISUP.py for ISUP.
Go through the readme and wiki, read the source code eventually. Then you will be able to develop the application you need with Python.

AmroAbdulla, 2022-07-02:
where can i find m3ua

closed 2022-12-08 by p1-bmu