P1sec/pycrate #221: how to replace asn1 enumerated name to raw value

zywj, 2023-02-19:
The result is like this below:
```
{
    "key": "enum_name"
}
```
But,  I want the result is like this:
```
{
    "key": "23"
}
```

p1-bmu, 2023-02-27:
Sorry, but I don't understand what you want to do. Where would the "23" value come from ?

zywj, 2023-02-27:
Sorry, I didn't describe it clearly.  
```
Test ::= SEQUENCE{
  userType UserType OPTIONAL,
}

UserType ::= ENUMERATED{
  admin (0),
  user (1)
}
END
```
I want get the result like this:
```
{
  userType: 1    
}
```
But now,  the result is this:
```
{
  userType: user 
}
```

p1-bmu, 2023-02-27:
In ASN.1, ENUMERATED values are represented by their string, not their index. So, there is currently no alternative value representation for those abject apart the string one. On the other side, the ENUM class has a `_get_index()` method which returns the index related to the value set. For example:
```
from pycrate_asn1dir.S1AP import *
cp = S1AP_IEs.CauseProtocol                                                                                                                                                                         
cp.set_val('semantic-error')
print(cp.get_val()) # 'semantic-error'
print(cp._get_index()) # 4
```

zywj, 2023-02-28:
Thanks for your valuable help

closed 2023-02-28 by zywj