P1sec/pycrate #222: ASN1ObjErr: InitiatingMessage.privateIEs: value out of size constraint, []

Onikauz, 2023-03-17:
Hello, 

I got an error when I create a NGAP PDU with a PrivateMessageIE.   
Just below the Traceback: 

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/dist-packages/pycrate-0.5.4-py3.8.egg/pycrate_asn1rt/asnobj.py", line 1352, in from_asn1
    self._safechk_bnd(self._val)
  File "/usr/local/lib/python3.8/dist-packages/pycrate-0.5.4-py3.8.egg/pycrate_asn1rt/asnobj_construct.py", line 94, in _safechk_bnd
    self._cont[val[0]]._safechk_bnd(val[1])
  File "/usr/local/lib/python3.8/dist-packages/pycrate-0.5.4-py3.8.egg/pycrate_asn1rt/asnobj_construct.py", line 776, in _safechk_bnd
    Obj._safechk_bnd(Obj._val)
  File "/usr/local/lib/python3.8/dist-packages/pycrate-0.5.4-py3.8.egg/pycrate_asn1rt/asnobj_ext.py", line 184, in _safechk_bnd
    self._get_val_obj(val[0])._safechk_bnd(val[1])
  File "/usr/local/lib/python3.8/dist-packages/pycrate-0.5.4-py3.8.egg/pycrate_asn1rt/asnobj_construct.py", line 776, in _safechk_bnd
    Obj._safechk_bnd(Obj._val)
  File "/usr/local/lib/python3.8/dist-packages/pycrate-0.5.4-py3.8.egg/pycrate_asn1rt/asnobj_construct.py", line 2645, in _safechk_bnd
    raise(ASN1ObjErr('{0}: value out of size constraint, {1!r}'\
pycrate_asn1rt.err.ASN1ObjErr: InitiatingMessage.privateIEs: value out of size constraint, []
```
Here a code to reproduce the issue:

```
>>> import pycrate_asn1dir.NGAP
>>> pdu = pycrate_asn1dir.NGAP.NGAP_PDU_Descriptions.NGAP_PDU
>>> asn1="""initiatingMessage : {
...   procedureCode 31,
...   criticality ignore,
...   value PrivateMessage: {
...     privateIEs {
...     }
...   }
... }""">>> pdu.from_asn1(asn1)
```

Thanks

p1-bmu, 2023-03-17:
As you can read from the error message, your `privateIEs` value is out of size constraint. This is because you provide actually a value of size 0, while the NGAP spec requires a value of size at least 1:
```
PrivateIE-Container {NGAP-PRIVATE-IES : IEsSetParam } ::= SEQUENCE (SIZE (1..maxPrivateIEs)) OF PrivateIE-Field {{IEsSetParam}}
```

p1-bmu, 2023-03-17:
You need to declare some specific structures, into `privateIEs` as part of the NGAP ASN.1 specification in the `pycrate_asn1dir` directory, then recompile the specification. Without this, you will only be able to set raw buffers into this sequence, in my recollection.

p1-bmu, 2023-03-22:
Here is an example. The prototype for the `privateIEs` object is:
```
In [10]: NGAP_PDU_Contents.PrivateMessage.get_proto()                                                                                                                                                        
Out[10]: 
('SEQUENCE',
 {
 privateIEs: ('SEQUENCE OF', ('SEQUENCE', {
  id: ('CHOICE', {
   local: 'INTEGER',
   global: 'OBJECT IDENTIFIER'
   }),
  criticality: 'ENUMERATED',
  value: ('OPEN_TYPE', {})
  }))
 })
```

With a minimum length of 1 for the SEQUENCE OF. So, here is a possible (dummy boolean) value for such a Private NGAP PDU:
```
PDU = NGAP_PDU_Descriptions.NGAP_PDU
v = ('initiatingMessage', {'procedureCode': 31, 'criticality': 'ignore', 'value': ('PrivateMessage', {'privateIEs': [{'id': ('local', 1), 'criticality': 'ignore', 'value': (TYPE_BOOL, False)}]})})
PDU.set_val(v)
print(PDU.to_asn1())
enc = PDU.to_aper()
```

Onikauz, 2023-03-23:
Hello, 

I have some other questions to understand well. 
To compile NGAP ASN1 specification I need to modify only .txt file or .asn file also ? 

I also test your example and it dosen't works, I got the same prototype, do I need to add this specification and compile ?
`
>>> import pycrate_asn1dir.NGAP
>>> pdu = pycrate_asn1dir.NGAP.NGAP_PDU_Descriptions.NGAP_PDU
>>> v = ('initiatingMessage', {'procedureCode': 31, 'criticality': 'ignore', 'value': ('PrivateMessage', {'privateIEs': [{'id': ('local', 1), 'criticality': 'ignore', 'value': (pycrate_asn1dir.NGAP.TYPE_BOOL, False)}]})})
>>> pdu.set_val(v)
ENUM._safechk_bnd: criticality, unable to retrieve a defined object
`

p1-bmu, 2023-03-23:
Other question => open another issue, but please ensure first you don't have enough information from the README and the wiki.

Regarding the example provided, what doesn't work exactly ? You say you "got the same prototype", what does it mean ?

closed 2023-09-08 by p1-bmu