P1sec/pycrate #226: Additionnal Information Request and MSNetCap encoding cannot be decoded

Marc-Egli, 2023-03-23:
I am trying to add a `MSNetCap` and a `AddInfoReq` field to a `EMMTrackingAreaUpdateRequest`

Here is a small snippet of code describing what I do : 

```py
from pycrate_mobile.NAS import *
from pycrate_csn1dir.ms_network_capability_value_part import ms_network_capability_value_part

# Valid MSNetCap Object
val = [1, 1, 1, 0, 1, 0, 1, 1, [1, 1, 1, 1, 1, 1], 0, 0, 0, 1, 1, 0, 1, 0, 0, 1, 0, 0, 0, 0, []]
MSNetCap = ms_network_capability_value_part.clone()
MSNetCap.set_val(val)
show(MSNetCap)

fields = {'MSNetCap' : val, 'AddInfoReq' : {'spare' : 0, 'CipherKey': 1}}
msg = EMMTrackingAreaUpdateRequest(val=fields)

# Valid message with MSNetCap field values set
show(msg)

Msg, err = parse_NAS_MT(unhexlify(msg.to_bytes().hex()))

if err != 0:
    print(err)
    
show(Msg)
```

The `parse_NAS_MT` function fails with error `96`, which means I am missing a mandatory field. I do not understand why as I generate the `MSNetCap` according to the specifications. I noticed in the wiki that there was an explanation about decoding and encoding `MSNetCap` fields. I tried using the values from there instead of my own values. The difference is that the value from the wiki which is: `[1, 1, 1, 0, 1, 0, 1, 1, [1, 1, 1, 1, 1, 1], 0, 0, 0, 1, 1, 0, 1, 0, 0]`, misses the last four single bit fields and the spare bit reference. To my surprise using the wiki value worked and the message was correctly re-encoded. I then tried the following value: `[1, 1, 1, 0, 1, 0, 1, 1, [1, 1, 1, 1, 1, 1], 0, 0, 0, 1, 1, 0, 1, 0, 0, 1, 0, 0, 0, 0, [0,0,0,0]]` which also worked but the decoding was wrong as shown below.

```
  <ms_network_capability_value_part (CSN1List): [
   <(CSN1Ref): <gea1_bits (CSN1Bit): 1>>
   <sm_capabilities_via_dedicated_channels (CSN1Bit): 1>
   <sm_capabilities_via_gprs_channels (CSN1Bit): 1>
   <ucs2_support (CSN1Bit): 0>
   <ss_screening_indicator (CSN1Bit): 1>
   <solsa_capability (CSN1Bit): 0>
   <revision_level_indicator (CSN1Bit): 1>
   <pfc_feature_mode (CSN1Bit): 1>
   <(CSN1Ref): <extended_gea_bits (CSN1List): [
    <gea_2 (CSN1Bit): 1>
    <gea_3 (CSN1Bit): 1>
    <gea_4 (CSN1Bit): 1>
    <gea_5 (CSN1Bit): 1>
    <gea_6 (CSN1Bit): 1>
    <gea_7 (CSN1Bit): 1>]>>
   <lcs_va_capability (CSN1Bit): 0>
   <ps_inter_rat_ho_from_geran_to_utran_iu_mode_capability (CSN1Bit): 0>
   <ps_inter_rat_ho_from_geran_to_e_utran_s1_mode_capability (CSN1Bit): 0>
   <emm_combined_procedures_capability (CSN1Bit): 1>
   <isr_support (CSN1Bit): 1>
   <srvcc_to_geran_utran_capability (CSN1Bit): 0>
   <epc_capability (CSN1Bit): 1>
   <nf_capability (CSN1Bit): 0>
   <geran_network_sharing_capability (CSN1Bit): 0>
   <user_plane_integrity_protection_support (CSN1Bit): 1>
   <gia_4 (CSN1Bit): 0>
   <gia_5 (CSN1Bit): 0>
   <gia_6 (CSN1Bit): 0>
   <gia_7 (CSN1Bit): 0>
   <(CSN1Ref): <spare_bits (CSN1Bit): [0,
   0,
   0]>>]>
```
The `spare_bits` field contains only 3 bits whereas in my `MSNetCap` field I specified 4 bits set to 0 in the `spare_bits` reference field.

Another observation I made is that removing the `AddinfoReq` will solve the issue and make the 4th bit reappear in the message.
I am still looking into the source code to find out why this happens.
I think there is an issue with overlapping fields.

Thanks in advance.

p1-bmu, 2023-03-23:
Thanks for your feedback. CSN.1 fields are quite painful to manage unfortunately... Below, we can see that the `get_len()` method for them are not exactly aligned with `get_bl()` (i.e. for getting the length in bits), as it returns the round length in bytes, omitting the trailing / padding bits required for the NAS IE to align.

```
In [49]: MSNetCap.set_val([1, 1, 1, 0, 1, 0, 1, 1, [1, 1, 1, 1, 1, 1], 0, 0, 0, 1, 1, 0, 1, 0, 0, 1, 0, 0, 0, 0, []])                                                                                        

In [50]: MSNetCap.get_len()                                                                                                                                                                                  
Out[50]: 3

In [51]: MSNetCap.get_bl()                                                                                                                                                                                   
Out[51]: 29

In [52]: MSNetCap.set_val([1, 1, 1, 0, 1, 0, 1, 1, [1, 1, 1, 1, 1, 1], 0, 0, 0, 1, 1, 0, 1, 0, 0, 1, 0, 0, 0, 0, [0, 0, 0]])                                                                                 

In [53]: MSNetCap.get_bl()                                                                                                                                                                                   
Out[53]: 32

In [54]: MSNetCap.get_len()                                                                                                                                                                                  
Out[54]: 4

In [55]: m = EMMTrackingAreaUpdateRequest(val={'MSNetCap': [1, 1, 1, 0, 1, 0, 1, 1, [1, 1, 1, 1, 1, 1], 0, 0, 0, 1, 1, 0, 1, 0, 0, 1, 0, 0, 0, 0, [0, 0, 0]], 'AddInfoReq': {'CipherKey': 1}})               

In [56]: m, e = parse_NAS_MO(m.to_bytes())                                                                                                                                                                   

In [57]: e                                                                                                                                                                                                   
Out[57]: 0
```

The workaround is to insert the padding at the end of the `MSNetCap` value explicitely. I need to check how easy it is to fix the `get_len()` method for the CSN1 objects.

Marc-Egli, 2023-03-24:
Thank you very much ! Works for me now.

closed 2023-03-24 by Marc-Egli