P1sec/pycrate #230: Question: conversion of 3GPP doc files to .txt

frankist, 2023-04-05:
Hi everyone,

There used to be `extract.py` scripts in the `pycrate_asn1dir/` NR folders that were used to convert the 3GPP .doc files into .asn files. However, these scripts have been removed in the latest pycrate releases. Is there a reason why?

I would like to compile the latest 3GPP NR TS rel17 docs, but without the `extract.py` script, I am not sure how to do it.

Thanks for the help. Very helpful project!

p1-bmu, 2023-04-06:
Hello,
I removed those script as I don't want to maintain them (sorry for this). You can download an older release of the project (e.g. 0.5.2) in order to retrieve them, or just `git checkout` to the tag.

p1-bmu, 2023-04-06:
By the way, I'll update all the 3GPP RAN specs to Rel.17 soon.

frankist, 2023-04-06:
I see. So, I guess it is a matter of time until the 0.5.2 scripts become outdated. In case, anyone else knows another project that has a similar .doc -> .asn conversion functionality, feel free to share it here.

Thanks for the help.

p1-bmu, 2023-04-07:
I just tagged a new 0.6.0 release after updating 4G and 5G RAN ASN.1 specs to Rel.17.

Regarding 3GPP specs conversion, you can download zip files from the 3GPP server, then script libreoffice in headless mode for .doc / .docx conversion to text (beware of non-ascii UTF-8 characters being sometimes present), and finally manage ASN.1 definitions extraction. There is also the https://getsi.org/ web page, which enables to download 3GPP specs in PDF format from the ETSI server, which you can then convert with `pdftotext`.

frankist, 2023-04-07:
Thanks very much for the release update and providing the steps for asn extraction. Hopefully, this will be of help to others as well.

closed 2023-04-21 by p1-bmu