P1sec/pycrate #232: Value mismatch while decoding the GTP hex stream

Sohel1999, 2023-04-13:
I am decoding a GTP hex stream from pcap.
In PCAP, the value of cause is 128. (cause: accepted accepted (128))
but while decoding the hex stream from that pcap, the cause value is 0. (Value: 0 (request accepted))
why values are different plz check it. 
I have attached the code and the pcap for your reference.

Pycrate version: 0.5.5
Python version: 3.8.5

Example:

`from binascii import *
from pycrate_mobile.TS29060_GTP import *
x="321100690000000112340000018008fe0e0110451f280011451c36737f451f2800800006f121000000008400228080211003000010810600000000830608080808000d0400000000000d04080808080000000000000000000000000000870011031b931f9396fefe74fffeff0019001414"
m = unhexlify(x)
Msg, Err = parse_GTP(m)
print(Msg[1][0].show())`

    ### Cause ###
     <Type : 1 (Cause)>
     ### Data ###
      <Resp : 1>
      <Reject : 0>
      <Value : 0 (Request accepted)>

Pcap:
[test2.zip](https://github.com/P1sec/pycrate/files/11220916/test2.zip)

Another Example:
where cause in pcap was 219 and while decoding it was 27.
`x = "32110013000000011234000001db0e018400088080210403000004"
m = unhexlify(x)
Msg, Err = parse_GTP(m)
print(Msg[1][0].show())`

    ### Cause ###
     <Type : 1 (Cause)>
     ### Data ###
      <Resp : 1>
      <Reject : 1>
      <Value : 27 (Missing or unknown APN)>

Screenshot:
[image]

p1-bmu, 2023-04-13:
Actually, there is nothing wrong here: pycrate manages the GTP Cause structure with subfields `Resp` (1 bit), `Reject` (1 bit) and `Value` (6 bits). So, in your decoded structures, having 
```
 ### Data ###
  <Resp : 1>
  <Reject : 0>
  <Value : 0 (Request accepted)>
```
is exactly the same as having an overall GTP Cause of value 0x80 (or 128). This is more an issue with wireshark which does not indicate the subfields' value, but only provides the overall Cause value.

Sohel1999, 2023-04-17:
Thank you for the quick response.

closed 2023-04-21 by p1-bmu