P1sec/pycrate #239: EMM EmergNumList field : Value not updated when using set_val

Marc-Egli, 2023-06-19:
Hello,

I have been using the `set_val` and `get_val` methods to modify field values in EMM messages. It is working flawlessly except for the `EmergNumList` IE. Whenever I try to set the `Len` value of an `EmergNum` item it does not get updated. I have tested on other fields and it works without any issues. I can for example modify the length of the `EmergNumList`.

Here is a code snippet to reproduce the issue

```py

from pycrate_mobile.NAS import *
from pycrate_mobile import *

val = {'EmergNumList': {0: {'spare': 0, 'ServiceCat': {'Police': 0, 'Ambulance': 1,  'Fire': 0, 'Marine': 0, 'Mountain': 1}, 'Num': b'\xa9\xa4\x0e\x89Y4\x84\x05'}}}

msg = EMMAttachAccept(val=val)

# Length value is correctly modified
print(msg['EmergNumList']['L'].get_val())
msg['EmergNumList']['L'].set_val(1)
print(msg['EmergNumList']['L'].get_val())

# Nothing happens
print(msg['EmergNumList']['EmergNumList'][0]['Len'].get_val())
msg['EmergNumList']['EmergNumList'][0]['Len'].set_val(4)
print(msg['EmergNumList']['EmergNumList'][0]['Len'].get_val())

```

I also tried to modify the `Num` field which also did not work.
I also thought that the issue might come from the `set_valauto` function defined at https://github.com/P1sec/pycrate/blob/47aaef8528d459e4df68a9854079c76679c7daec/pycrate_mobile/TS24008_IE.py#L1129
but I do not think this is the root of the problem.

Thank you for your help.

p1-bmu, 2023-06-19:
Thanks for reporting.
Commit https://github.com/P1sec/pycrate/commit/e20b5013a33ceea09a935ba25685ae12a2d75b3f fixes the issue.

Marc-Egli, 2023-06-20:
Thank you very much !

closed 2023-06-20 by Marc-Egli