P1sec/pycrate #241: Error when decoding Kerberos from BER

joelcawthorne, 2023-07-05:
When trying to decode Kerberos BER data, I'm running into the follow error:

`ASN1ObjErr: PrincipalName.name-string._item_: GeneralString value out of constraint, 'testuser'`

This coming from `site-packages/pycrate_asn1rt/asnobj.py:340`

Despite this error, the subsequent call to `GLOBAL.MOD['KerberosV5Spec2']['KDC-REQ']()` till produces all of the correct data from the provided bytes.

Here are the sample bytes I'm using:

`3081b9a103020105a20302010aa30e300c300aa10402020095a2020400a4819c308199a00703050040800000a1153013a003020101a10c300a1b087465737475736572a20c1b0a4d595445534c412e4a43a31f301da003020102a11630141b066b72627467741b0a4d595445534c412e4a43a511180f32303233303632343130303831365aa611180f32303233303633303130303831365aa706020420dc639da81a301802011202011102011402011302011002011702011902011a`

joelcawthorne, 2023-07-06:
I read in a similar issue that setting `ASN1Obj._SAFE_BND = False` works, and so I tested this setting and it resolved my issues.

closed 2023-07-06 by joelcawthorne

p1-bmu, 2023-07-17:
Hello,
Thanks for your feedback. Actually, the object `name-string` contains `GenericString` which is actually an ISO2022 character string. In pycrate, it is linked to the japanese ISO2022 codec: https://github.com/P1sec/pycrate/blob/6fd12f67ce44bfa5e8374447f5e4311f349082e6/pycrate_asn1rt/asnobj_str.py#L2865

But this is only a partial ISO2022 support, which fails here with a standard ascii string unfortunately. Therefore, disabling the constraint checking on the character set seems an appropriate solution here. I verified that the re-encoding provides the same buffer as the one consumed in that case.