P1sec/pycrate #242: ASN.1 RRCLTE Error : `get_at().get_val() yields different result as `get_val_at()`

Marc-Egli, 2023-07-28:
Hello,

I am currently working with RRCLTE and I noticed the following error

```py
from pycrate_asn1dir import RRCLTE
from pycrate_asn1rt import *

target_field_path = ['message', 'c1', 'rrcConnectionReconfiguration', 'criticalExtensions', 'c1', 'rrcConnectionReconfiguration-r8', 'nonCriticalExtension', 'nonCriticalExtension', 'nonCriticalExtension', 'nonCriticalExtension', 'nonCriticalExtension', 'nonCriticalExtension', 'nonCriticalExtension', 'nonCriticalExtension', 'nonCriticalExtension', 'sCellGroupToAddModList-r15', 0, 'sCellToAddModList-r15', 0, 'radioResourceConfigDedicatedSCell-r13', 'physicalConfigDedicatedSCell-r10', 'epdcch-Config-r11', 'config-r11', 'setup', 'setConfigToAddModList-r11', 0, 'resourceBlockAssignment-r11', 'resourceBlockAssignment-r11']

msg_bytes = b'&\x00\xa4\xa0\x82\x10H%\ty\x82`\x00\xb0\x84g\x01k\xb1\x06p\x14\xe1\x95\x80\x08\x00'

packet = RRCLTE.EUTRA_RRC_Definitions.DL_DCCH_Message

packet.from_uper(msg_bytes)

print(packet.get_val_at(target_field_path)) # Gives (10, 4)
print(packet.get_at(target_field_path).get_val()) # Gives (11, 4)

print(packet.get_at(target_field_path).to_uper()) # Gives b'\x02\x80'

assert packet.get_val_at(target_field_path) == packet.get_at(target_field_path).get_val()

```

In this smal PoC the values of the field `resourceBlockAssignment-r11` are different depending on how they are querried. From my understanding `4` is the length of the bit string and `10` or `11`in this case is the value of it. It seems that the value is of by one depending on the method chosen to query.

`b'\x02\x80'` is `0000001010000000` in binary when encoding in uper, where the first 6 bits are used to indicate the length of the bit string (from the specification `resourceBlockAssignment-r11` has a length between `4-38` which means 6 bits are needed to represent it. Then because the value length is 4 we have 4 bits for the value which in this case is `10` (`1010` in binary). The remaining zeroes are padding bits.

I then inspected the packet and the uper value of the target field and it seems that `10` is the correct value.
I will continue to investigate this also on my side.

Thank you in advance.

p1-bmu, 2023-07-28:
Thanks for your feedback.

This is an expected behaviour actually. The `get_val_at()` method lets you navigate through the whole object's value, while `get_at()` navigates within the object's structure only.

In SEQUENCE OF or SET OF objects, the child object is not deduplicated and the same object is reused to decode multiple buffers and values sequentially. So when you use `get_at()`, you select the same child object of the SEQUENCE OF or SET OF whatever index you provide (0, 1, 2...), while when you use `get_val_at()` you select distinct values according to the index within the SEQUENCE OF or SET OF.
What happens here is that when you select the [`setConfigToAddModList-r11`, 0] component with `get_at()`, you only get the object `EPDCCH-SetConfig-r11`, which retains the last value it has decoded; with object selection, such component's index is meaningless: index 0, 1, 2... within a SEQUENCE OF or SET OF will always return the same child component. But when you select the [`setConfigToAddModList-r11`, 0] value with `get_val_at()`, you properly get the 1st value decoded for this component.

All in all, when you want to work on objects solely, always use `get_at()`; and when you want to work on values, always use `get_val_at()`.

Marc-Egli, 2023-07-28:
Thank you very much for your quick and clear answer !
I didn't realize that they were different because of the naming. 
My reasoning is that `get_val_at` was a combination of `get_at` and `get_val` and that both could be used to work with values.

Does this mean that there is also a difference between `set_val` and `set_val_at`. In my PoC is use `set_val` but I could also use `set_val_at` with an empty path. Would there be any difference ?

p1-bmu, 2023-07-28:
`set_val()` sets the value directly on the object it is called on, while `set_val_at()` enables you to set a sub-value of the object you are calling it on. When you use `set_val_at()`, it generally means the object has already an overall value set, that you will traverse to set the sub-value at the appropriate path.
Using `set_val_at()` with an empty path is identical to using `set_val()`.

closed 2023-09-08 by p1-bmu