P1sec/pycrate #243: ASN1 encoding error when ... extension

luiscpgh, 2023-08-14:
Hi,

I encountered an issue encoding a message containing IEs in the "..." extension (sorry, I am not sure what this extension is actually called)...

Decoding works great. Encoding is the problem.

Here is a minimal example:

```
from pycrate_asn1dir import RRCNR
from binascii import unhexlify, hexlify

# decode
rrc = RRCNR.NR_RRC_Definitions.UE_MRDC_Capability
hexstr = '''a107c058161601c41000000122200402c2340410b003a030a74ed8e7c310000800040202008008'''
unhex = unhexlify(hexstr)
rrc.from_uper(unhexlify(hexstr))
decoded  = rrc()
decodedMsg = rrc.to_asn1()
print(decodedMsg)

# now encode it
en_rrc = RRCNR.NR_RRC_Definitions.UE_MRDC_Capability
en_rrc.from_asn1(decodedMsg) # error:  unknown extension, 'supportedBandCombinationList-v1540 {\n...
en_unhex = en_rrc.to_uper()
en_hex = bytes.hex(en_unhex)

print('encoded {}'.format(en_hex))
```

In the example above, I think the issue is that the supportedBandCombinationList-v1540 IE is past the "..." extension in the 3GPP spec:
```
RF-ParametersMRDC ::=                   SEQUENCE {
    supportedBandCombinationList            BandCombinationList                             OPTIONAL,
    appliedFreqBandListFilter               FreqBandList                                    OPTIONAL,
    ...,
    [[
    srs-SwitchingTimeRequested              ENUMERATED {true}                               OPTIONAL,
    supportedBandCombinationList-v1540      BandCombinationList-v1540                       OPTIONAL
    ]],
```
Any tips with how to tell the encoder that the IE is past the "..." extension?

p1-bmu, 2023-08-28:
Actually here, this is not an encoding but a decoding issue. The error happens when you try to decode the textual string corresponding to the value in ASN.1 notation, into the object structure.
There is no issue when you set the direct value into the object:
```
en_rrc.set_val(decoded)
```

I'll need to investigate what is the exact issue with decoding the ASN.1 value string.

p1-bmu, 2023-08-28:
We have grouped extensions in the sub-object _RF-ParametersMRDC_:
```
RF-ParametersMRDC ::=                   SEQUENCE {
    supportedBandCombinationList            BandCombinationList                             OPTIONAL,
    appliedFreqBandListFilter               FreqBandList                                    OPTIONAL,
    ...,
    [[
    srs-SwitchingTimeRequested              ENUMERATED {true}                               OPTIONAL,
    supportedBandCombinationList-v1540      BandCombinationList-v1540                       OPTIONAL
    ]],
```

And the group components are all _OPTIONAL_, plus the _srs-SwitchingTimeRequested_ can only have one value forced to _True_. Hence, we may have additionally an issue here with canonical PER. As a first remark, I see the UPER re-encoding from pycrate differs from the provided buffer.

@luiscpgh : did the buffer has been observe from a commercial handset implementation ?

luiscpgh, 2023-08-30:
@p1-bmu : Thank you for looking into this. Yes, the message is sent by a OnePlus N200 phone.