P1sec/pycrate #247: 5G NAS UE policy protocol TS 24.501_UEPOL RouteSelectDesc from_bytes inconsistent with to_bytes

maddenj-ie, 2023-11-02:
Hi,
I've been experimenting with your library with a view to using it to test UEPOL related functionality.

I've been testing the from_bytes & to_bytes on various classes.

The following pytest fails.

`
from pycrate_mobile.TS24526_UEPOL import RouteSelectDesc
from pycrate_core.repr import show

class TestRouteSelectDesc:

    def test_route_select_desc_from_bytes_to_obj(self):

        route_select_desc_orig = RouteSelectDesc(val={'Precedence': 1,
                                                      'Cont': [{'Type': 16, 'Value': {'Value': 1}}]})
        show(route_select_desc_orig)
        route_select_desc = RouteSelectDesc()
        route_select_desc.from_bytes(route_select_desc_orig.to_bytes())
        show(route_select_desc)
        assert route_select_desc.get_val() == route_select_desc_orig.get_val()
`

Thanks for your help.

maddenj-ie, 2023-11-02:
Investigating further

https://github.com/P1sec/pycrate/blob/7da0c691ba56c8ca257834af75e87baed9d47f3c/pycrate_mobile/TS24526_UEPOL.py#L393

I believe this should be:

```python
self[3].set_blauto(lambda: self[2].get_val() << 3)
```

I suspect that there may be other occurrences of the same issue in the same file

p1-bmu, 2023-11-08:
Thanks for your feedback. There are effectively some issues in this 5G NAS UE Policy implementation... I committed a first series of fix here: https://github.com/P1sec/pycrate/commit/788ed0f72e7ba41efea60790f1c4f7242c592000, what should address your case.
Actually, I have never tested this part of the library with real world signalling data. Therefore, in case you have some examples of UE Policy NAS exchanges that pycrate could test against, or if you know where such examples could be found: please tell me, that would be awesome !

closed 2023-11-08 by p1-bmu