P1sec/pycrate #248: FG UE Pol Manage Command fromJson

maddenj-ie, 2023-11-17:
Hi,

I'm experimenting with the FG UE Policy Management Command to/from json functionality using the pytest below.

```python
from pycrate_core.repr import show
from pycrate_mobile.TS24501_UEPOL import FGUEPOLManageUEPolicyCommand

class TestManageUePolicyCommandJson:

    def test_to_from_json(self):
        command_obj = FGUEPOLManageUEPolicyCommand(val={
            'UEPolSectionList': [
                {
                    'PLMN': '27201',
                    'Cont': [
                        {
                            'UPSC': 1,
                            'Cont': [
                                {
                                    'Type': 1,
                                    'Cont': [
                                        {
                                            'Precedence': 1,
                                            'TrafficDesc': [
                                                {
                                                    'Type': 16,
                                                    'Value':
                                                        {
                                                            'Addr': b'aaaa',
                                                            'Mask': b'BBBB'
                                                        }
                                                }
                                            ],
                                            'RouteSelectDescList': [
                                                {
                                                    'Precedence': 1,
                                                    'Cont': [
                                                        {
                                                            'Type': 16,
                                                            'Value':
                                                                {
                                                                    'Value': 1
                                                                }
                                                        }
                                                    ]
                                                }
                                            ]
                                        }
                                    ]
                                }
                            ]
                        }
                    ]
                }
            ]
        })
        show(command_obj)
        generated_json = command_obj.to_json()
        print("generated JSON : " + generated_json)

        parsed_json_command = FGUEPOLManageUEPolicyCommand()
        parsed_json_command.from_json(generated_json)
        show(parsed_json_command)
```

Using the show command, the parsed command obj is as follows:

```
### 5GUEPOLManageUEPolicyCommand ###
 ### 5GUEPOLHeader ###
  <PTI : 0>
  <Type : 1 (MANAGE UE POLICY COMMAND message)>
 ### UEPolSectionList ###
  <L : 35>
  <V : 0x000700000000020000>
```

The toJson works as I'd expect but the fromJson does not.

This may be down to user error. ;-)

I'd appreciate any feedback.  Thanks.

p1-bmu, 2023-11-23:
Thank you for the bug report.
There is a custom routine to set specific IEs in NAS Layer 3 messages, and I guess I did not extend for proper JSON support; I think a good starting point for a patch would be here:
https://github.com/P1sec/pycrate/blob/1824d90cd4815ccdf4070e76338009aa1a1e8097/pycrate_mobile/TS24007.py#L250

I'll check what I can do to fix this in the following weeks.

mitshell, 2024-03-01:
A new repository has been setup, including new fixes. This repo will keep being maintained : https://github.com/pycrate-org/pycrate.
Please checkout the latest 0.7.1 version which provides extended JSON support for mobile NAS messages.