P1sec/pycrate #249: Parsing `RRLP_Components.LocationError` gives wrong results

matan1008, 2023-11-30:
When parsing LocationError in RRLP component, the result might be wrong, for example:

```
from pycrate_mobile.TS44018_RR import *
error = RRLP_Components.LocationError
error.from_uper(b'\x04\x99\x02@\x00')
print(error())
```

Gives `{'locErrorReason': 'notEnoughSats'}`, but according to wireshark the parsing is 
[image]

mitshell, 2024-03-01:
RRLP is encoded is unaligned PER, which means the encoding is not aligned on byte-boundary. Hence, the buffer you copy from wireshark may not correspond to the exact PDU part you are looking for.
Here, if you offset your buffer from 7 bits, the decoding of this RRLP object will lead to the expected value:
```
{'locErrorReason': 'gpsAssDataMissing', 'additionalAssistanceData': {'gpsAssistanceData': b' \x00'}}
```

matan1008, 2024-03-02:
Thanks a lot!

closed 2024-03-02 by matan1008