P1sec/pycrate #256: Support ASN.1 for E2AP?

Yi-Hsueh-Tsai, 2024-01-24:
Would you support ASN.1 for E2AP in the near future?

mitshell, 2024-03-01:
Please note a new repository has been setup, including new fixes. This repo will keep being maintained : https://github.com/pycrate-org/pycrate.

For supporting E2AP, we need a freely available ASN.1 definition of the specification. I am not sure this is currently the case with O-RAN specifications, but this may be easier to get if it is released by the ETSI in PDF. Could you please indicate where you get your definition of E2AP ?

Yi-Hsueh-Tsai, 2024-03-02:
Dear all

    You can download O-RAN specifications regarding E2AP protocol and ASN.1
from the official website and github:
   https://orandownloadsweb.azurewebsites.net/specifications

https://github.com/o-ran-sc/ric-plt-e2/tree/master/RIC-E2-TERMINATION/3rdparty/asnTextFiles

Best Regards,

Lucas

Yi-Hsueh-Tsai, 2024-03-02:
[O-RAN.WG3.E2AP-R003-v05.00.docx](https://github.com/P1sec/pycrate/files/14467906/O-RAN.WG3.E2AP-R003-v05.00.docx)
[O-RAN.WG3.E2GAP-R003-v05.00.docx](https://github.com/P1sec/pycrate/files/14467908/O-RAN.WG3.E2GAP-R003-v05.00.docx)

mitshell, 2024-03-02:
E2AP support has been added in 0.7.2. See https://github.com/pycrate-org/pycrate